OpenCPN/OpenCPN #246: c++internal compiler error when building on Ubuntu 14.04

Gnimuc, 2015-02-01:
when I run `make`, an error interupted the process:
```
[ 12%] Built target NMEA0183
Scanning dependencies of target S57ENC
[ 12%] Building CXX object CMakeFiles/S57ENC.dir/src/TexFont.cpp.o
[ 12%] Building CXX object CMakeFiles/S57ENC.dir/src/s52cnsy.cpp.o
[ 13%] Building CXX object CMakeFiles/S57ENC.dir/src/s52plib.cpp.o
[ 13%] Building CXX object CMakeFiles/S57ENC.dir/src/s52utils.cpp.o
[ 13%] Building CXX object CMakeFiles/S57ENC.dir/src/s57chart.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/S57ENC.dir/src/s57chart.cpp.o] Error 4
make[1]: *** [CMakeFiles/S57ENC.dir/all] Error 2
make: *** [all] Error 2
```

I build it on a brand new ubuntu14.04 following the instruction here:http://opencpn.org/ocpn/compiling_source_linux 

is the instruction outdated?

Gnimuc, 2015-02-01:
I have searched `file:///usr/share/doc/gcc-4.8/`, but there is not a file called README.Bugs.

nohal, 2015-02-01:
The instructions are up to date and we build on Ubuntu 14.04 all the time (i386, amd64 and armhf). There seems to be something wrong with your particular machine.

closed 2015-02-01 by nohal