OpenCPN/OpenCPN #261: crash after then add chart directory

seaside82, 2015-03-23:
hi 
opencpn shutdown when i click to Options (with left click) after then i add chart directory. i use without add chart directory many times i can enter to Options but its not possible after then add chart directory. console says that.
Note my laptop is Toshiba Satellite 300-20F / Operating system is linux "Pardus" on debian based (here: http://distrowatch.com/table.php?distribution=pardus but they do not yet announce their last distro there which i use now its : http://www.parduslinux.org/pardus-topluluk-surumu-indir/ - 3.0 KARARLI - 32 bit) may distro repos addressess is ftp://community.pardus.org.tr/topluluk/anahavuz/  

Probably in 12 hours i will sail, i have not much time. Thanks 
 
ruhum11@ruhum11:~/OpenCPN/build$ sudo make install
[sudo] password for ruhum11: 
/usr/bin/cmake -H/home/ruhum11/OpenCPN -B/home/ruhum11/OpenCPN/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/ruhum11/OpenCPN/build/CMakeFiles /home/ruhum11/OpenCPN/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/ruhum11/OpenCPN/build'
make -f CMakeFiles/GARMINHOST.dir/build.make CMakeFiles/GARMINHOST.dir/depend
make[2]: Entering directory '/home/ruhum11/OpenCPN/build'
cd /home/ruhum11/OpenCPN/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/ruhum11/OpenCPN /home/ruhum11/OpenCPN /home/ruhum11/OpenCPN/build /home/ruhum11/OpenCPN/build /home/ruhum11/OpenCPN/build/CMakeFiles/GARMINHOST.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/ruhum11/OpenCPN/build'
make -f CMakeFiles/GARMINHOST.dir/build.make CMakeFiles/GARMINHOST.dir/build
make[2]: Entering directory '/home/ruhum11/OpenCPN/build'
make[2]: Nothing to be done for 'CMakeFiles/GARMINHOST.dir/build'.
make[2]: Leaving directory '/home/ruhum11/OpenCPN/build'
/usr/bin/cmake -E cmake_progress_report /home/ruhum11/OpenCPN/build/CMakeFiles  1 2 3 4
[  4%] Built target GARMINHOST
make -f CMakeFiles/NMEA0183.dir/build.make CMakeFiles/NMEA0183.dir/depend
make[2]: Entering directory '/home/ruhum11/OpenCPN/build'
cd /home/ruhum11/OpenCPN/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/ruhum11/OpenCPN /home/ruhum11/OpenCPN /home/ruhum11/OpenCPN/build /home/ruhum11/OpenCPN/build /home/ruhum11/OpenCPN/build/CMakeFiles/NMEA0183.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/ruhum11/OpenCPN/build'
make -f CMakeFiles/NMEA0183.dir/build.make CMakeFiles/NMEA0183.dir/build
make[2]: Entering directory '/home/ruhum11/OpenCPN/build'
make[2]: Nothing to be done for 'CMakeFiles/NMEA0183.dir/build'.
make[2]: Leaving directory '/home/ruhum11/OpenCPN/build'
/usr/bin/cmake -E cmake_progress_report /home/ruhum11/OpenCPN/build/CMakeFiles  5 6 7 8 9 10
[ 12%] Built target NMEA0183
make -f CMakeFiles/S57ENC.dir/build.make CMakeFiles/S57ENC.dir/depend
make[2]: Entering directory '/home/ruhum11/OpenCPN/build'
cd /home/ruhum11/OpenCPN/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/ruhum11/OpenCPN /home/ruhum11/OpenCPN /home/ruhum11/OpenCPN/build /home/ruhum11/OpenCPN/build /home/ruhum11/OpenCPN/build/CMakeFiles/S57ENC.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/ruhum11/OpenCPN/build'
make -f CMakeFiles/S57ENC.dir/build.make CMakeFiles/S57ENC.dir/build
make[2]: Entering directory '/home/ruhum11/OpenCPN/build'
make[2]: Nothing to be done for 'CMakeFiles/S57ENC.dir/build'.
make[2]: Leaving directory '/home/ruhum11/OpenCPN/build'
/usr/bin/cmake -E cmake_progress_report /home/ruhum11/OpenCPN/build/CMakeFiles  11 12 13 14 15 16 17 18 19 20 21 22
[ 26%] Built target S57ENC
make -f CMakeFiles/SYMBOLS.dir/build.make CMakeFiles/SYMBOLS.dir/depend
make[2]: Entering directory '/home/ruhum11/OpenCPN/build'                                                                                                    
cd /home/ruhum11/OpenCPN/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/ruhum11/OpenCPN /home/ruhum11/OpenCPN /home/ruhum11/OpenCPN/build /home/ruhum11/OpenCPN/build /home/ruhum11/OpenCPN/build/CMakeFiles/SYMBOLS.dir/DependInfo.cmake --color=                                                       
make[2]: Leaving directory '/home/ruhum11/OpenCPN/build'                                                                                                     
make -f CMakeFiles/SYMBOLS.dir/build.make CMakeFiles/SYMBOLS.dir/build                                                                                       
make[2]: Entering directory '/home/ruhum11/OpenCPN/build'                                                                                                    
make[2]: Nothing to be done for 'CMakeFiles/SYMBOLS.dir/build'.
make[2]: Leaving directory '/home/ruhum11/OpenCPN/build'
/usr/bin/cmake -E cmake_progress_report /home/ruhum11/OpenCPN/build/CMakeFiles 
[ 26%] Built target SYMBOLS
make -f CMakeFiles/TEXCMP.dir/build.make CMakeFiles/TEXCMP.dir/depend
make[2]: Entering directory '/home/ruhum11/OpenCPN/build'
cd /home/ruhum11/OpenCPN/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/ruhum11/OpenCPN /home/ruhum11/OpenCPN /home/ruhum11/OpenCPN/build /home/ruhum11/OpenCPN/build /home/ruhum11/OpenCPN/build/CMakeFiles/TEXCMP.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/ruhum11/OpenCPN/build'
make -f CMakeFiles/TEXCMP.dir/build.make CMakeFiles/TEXCMP.dir/build
make[2]: Entering directory '/home/ruhum11/OpenCPN/build'
make[2]: Nothing to be done for 'CMakeFiles/TEXCMP.dir/build'.
make[2]: Leaving directory '/home/ruhum11/OpenCPN/build'
/usr/bin/cmake -E cmake_progress_report /home/ruhum11/OpenCPN/build/CMakeFiles  23 24 25
[ 30%] Built target TEXCMP
make -f CMakeFiles/i18n.dir/build.make CMakeFiles/i18n.dir/depend
make[2]: Entering directory '/home/ruhum11/OpenCPN/build'
cd /home/ruhum11/OpenCPN/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/ruhum11/OpenCPN /home/ruhum11/OpenCPN /home/ruhum11/OpenCPN/build /home/ruhum11/OpenCPN/build /home/ruhum11/OpenCPN/build/CMakeFiles/i18n.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/ruhum11/OpenCPN/build'
make -f CMakeFiles/i18n.dir/build.make CMakeFiles/i18n.dir/build
make[2]: Entering directory '/home/ruhum11/OpenCPN/build'
/usr/bin/cmake -E cmake_progress_report /home/ruhum11/OpenCPN/build/CMakeFiles 
[ 30%] i18n: Done.
[... 183 lines cut ...]
0xb7606330 in wxEvtHandler::ProcessEvent(wxEvent&) at ??:0
0xb7605eef in wxEvtHandler::DoTryChain(wxEvent&) at ??:0
0xb76061eb in wxEvtHandler::ProcessEventLocally(wxEvent&) at ??:0
0xb7606243 in wxEvtHandler::ProcessEvent(wxEvent&) at ??:0
0xb71f4696 in wxWindowBase::TryAfter(wxEvent&) at ??:0
0xb76062fb in wxEvtHandler::ProcessEvent(wxEvent&) at ??:0
0xb76074cc in wxEvtHandler::ProcessPendingEvents() at ??:0
0xb74596d8 in wxAppConsoleBase::ProcessPendingEvents() at ??:0
0xb6f790c3 in wxApp::DoIdle() at ??:0
0xb6f7923c in  at ??:0
0xb697b500 in  at ??:0
0xb697ecb3 in g_main_context_dispatch at ??:0
0xb697f0c9 in  at ??:0
0xb697f479 in g_main_loop_run at ??:0
0xb60661b5 in gtk_main at ??:0
0xb6f9bc95 in wxGUIEventLoop::DoRun() at ??:0
0xb749f2f9 in wxEventLoopBase::Run() at ??:0
0xb74579af in wxAppConsoleBase::MainLoop() at ??:0
0xb7454f50 in wxAppConsoleBase::OnRun() at ??:0
0xb708ef2f in wxAppBase::OnRun() at ??:0
0xb74f9c68 in wxEntry(int&, wchar_t**) at ??:0
0xb74f9d33 in wxEntry(int&, char**) at ??:0
0x815c876 in main at chart1.cpp:819
İptal edildi
ruhum11@ruhum11:~/OpenCPN/build$

transmitterdan, 2015-03-23:
It looks like you are building against wxWidgets V3. I don't think that is allowed.

seaside82, 2015-03-23:
It is default on my operating system. I dont install anything except opencpn and i just formatted my laptop yesterday.

nohal, 2015-03-23:
OpenCPN 4.0 is not tested with WX3 on Linux, while it builds and basically runs, you may experience all sorts of problems. If you want a fast solution, use WX2.8

seaside82, 2015-03-23:
Nohal I have not much experiance. This compiler is default on my system. You mean compile with wx2.8. what I have to do? Which packages I have to remove and install.

nohal, 2015-03-23:
wxWidgets is not a compiler, it is a library we use for the GUI.
My Turkish is not exactly fluent, but I suppose Pardus 3.0 is based on Debian Jessie, so the key to your efforts is having `libwxgtk2.8-dev` and not `libwxgtk3.0-dev` installed.
wx2.8 was removed from Jessie, so you need to use the packages from Sid for now. To do that, add 
```
deb ftp://ftp.debian.org/debian/ sid main
deb-src ftp://ftp.debian.org/debian/ sid main
```
To your `/etc/apt/sources.list` and run `sudo apt-get update`.
Then remove the wx3.0 development packages using `sudo apt-get remove libwxbase3.0-dev libwxgtk3.0-dev wx3.0-headers` and install the 2.8 ones using `sudo apt-get install libwxgtk2.8-dev`

Finally clean your build directory completely and run the standard `cmake .. && make`

To check what you have on your system run `dpkg -l '*wx*'`, it should look something like this on a Jessie based machine used ofr OpenCPN development (note that while I have wx3.0 runtime installed, there are no development packages but for 2.8):
```
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                              Version               Architecture          Description
+++-=================================-=====================-=====================-========================================================================
ii  codeblocks-libwxcontrib0          13.12svn10111-1       amd64                 Code::Blocks shared libraries for wxContribItems
un  libalien-wxwidgets-perl           <none>                <none>                (no description available)
un  libgl1-mesa-swx11-dev             <none>                <none>                (no description available)
ii  libwxbase2.8-0:amd64              2.8.12.1+dfsg2-2      amd64                 wxBase library (runtime) - non-GUI support classes of wxWidgets toolkit
ii  libwxbase2.8-dev                  2.8.12.1+dfsg2-2      amd64                 wxBase library (development) - non-GUI support classes of wxWidgets tool
ii  libwxbase3.0-0:amd64              3.0.2-1+b1            amd64                 wxBase library (runtime) - non-GUI support classes of wxWidgets toolkit
ii  libwxgtk2.8-0:amd64               2.8.12.1+dfsg2-2      amd64                 wxWidgets Cross-platform C++ GUI toolkit (GTK+ runtime)
ii  libwxgtk2.8-dev                   2.8.12.1+dfsg2-2      amd64                 wxWidgets Cross-platform C++ GUI toolkit (GTK+ development)
ii  libwxgtk3.0-0:amd64               3.0.2-1+b1            amd64                 wxWidgets Cross-platform C++ GUI toolkit (GTK+ runtime)
ii  libwxsmithlib0                    13.12svn10111-1       amd64                 wxSmith shared library (Code::Blocks plugin for RAD GUI editing)
ii  wx-common                         3.0.2-1+b1            amd64                 wxWidgets Cross-platform C++ GUI toolkit (common support files)
un  wx-i18n                           <none>                <none>                (no description available)
un  wx2.8-doc                         <none>                <none>                (no description available)
ii  wx2.8-headers                     2.8.12.1+dfsg2-2      amd64                 wxWidgets Cross-platform C++ GUI toolkit (header files)
ii  wx2.8-i18n                        2.8.12.1+dfsg2-2      all                   wxWidgets Cross-platform C++ GUI toolkit (i18n support)
```

seaside82, 2015-03-23:
nohal
thank you its work now only it does not show ENC Text even if i click or not. 
Omer

nohal, 2015-03-23:
Omer...
Well, it seems that now it's time to read the manual.

P.

closed 2015-03-23 by nohal

seaside82, 2015-03-23:
:) you are too quick thanx, i have to read, but it difficult to do on board because to much job here. i spend of all of my rest time to check and learn what happen here on opencpn and linux. i will be beter day by day.
nice to hear your voice quick

seaside82, 2015-03-24:
thanx i fix it :)