OpenCPN/OpenCPN #299: 4.0.0 will not install on XP

tracyandgiles, 2015-06-03:
Attempting to install 4.0.0 on XP gives the error message:
"Your Windows XP Service Pack version is not supported (it's too old)! You need at least Windows XP SP3."

However, SP3 is installed.  After inspecting NSIS.template.in, it appears that line 948:
${AndIf} ${AtMostServicePack} 2
should be:
${AndIf} ${AtLeastServicePack} 3

fixed, built, tested, commit and pushed here:
https://github.com/tracyandgiles/OpenCPN/commit/3b74a9989c916366117bd2b3bc3a37aba4edaebe

nohal, 2015-06-03:
Are you sure?
Your logic will IMO trigger the error message for XPSP3 but not for older SPs
And if it was impossible to install OpenCPN 4.0 on XP, we would probably hear about it from those thousands of downloaders (Or experienced it on some of the testing machines we have)

tracyandgiles, 2015-06-03:
You are right.  I just checked and the machine in question was xp 64-bit with sp2 (not sp3).  so I broke the logic for everybody else.  I made the incorrect assumption that the machine was running the "latest" but for xp64 that is sp2 (not sp3).

closed 2015-06-03 by tracyandgiles