OpenCPN/OpenCPN #376: Use a fleshed out and supported coding style

versable, 2015-09-08:
As the code rarely follows the proposed style. And the style guideline is rather short (http://opencpn.org/ocpn/code-style). I would move to a well established code style that supports linters, autoformatters, IDE plugins, etc...

A great tool that I regularly use is clang-format, which formats any file I edit on the fly. ( IDE set to call clang-format on save ). It supports the following formats: LLVM, Google, Chromium, Mozilla, WebKit.

It would be a good step to make the code a bit cleaner, and the effort is minimal ( except merging current pull request would be time consuming )

versable, 2015-09-08:
Here is a guide on how to use clang-format in Visual Studio: http://visualstudioextensions.vlasovstudio.com/2013/11/05/format-c-code-intelligently-in-visual-studio-with-clang-format/

nohal, 2015-09-08:
versable...
New code more or less follows the style, of course nobody ever reformatted the legacy codebase
IMO If to be done, we should first announce it is going to happen, set the enforcing policy, improve the documentation, finish and merge the big pulls. Then do the reformatting, not the other way around
As almost nobody of us uses VS, are you aware of integrations for clang-format into C::B and KDevelop?

Pavel

versable, 2015-09-08:
For C::B I'd use Scripting Commands (http://wiki.codeblocks.org/index.php?title=Scripting_commands)
> [[GetEditorManager().SaveActive();]] cmd.exe /c "timeout" "1" && "$(CODEBLOCKS)\sdk\LLVM\bin\clang-format.exe"

I'm sure there is something similar in KDevelop

Personally I use sublime, which has a working plugin for clang-format.

canboat, 2015-12-21:
FWIW -- I'm going to use @versable 's idea of using `clang-format` with the Google style in br24radar_pi. My .clang-format is:

```
   BasedOnStyle:                                   Google
   PenaltyBreakString:                             999999999
   ColumnLimit:                                    132
```

The latter two are my solution to keep clang-format from breaking up wxT() and _() macro content.

rgleason, 2021-04-23:
Move to "Discussions" ?  Show & Tell topic?

closed 2021-04-23 by bdbcat