OpenCPN/OpenCPN #544: compilation instructions need updating?

jevonlongdell, 2016-01-04:
Hi I'm using ubuntu 15.10 and I just tried to compile the source code using instructions at http://opencpn.org/ocpn/compiling_source_linux but these didn't work

It fails compiling a plugin. (see error message below)
According to https://github.com/nohal/chartdldr_pi it needs be be built "out of treee" I tried the instrions there but that failed too. (see error #2)

Error #1:
/home/XXX/progs/OpenCPN/plugins/chartdldr_pi/src/chartdldr_pi.cpp: In member function ‘virtual int chartdldr_pi::GetPlugInVersionMajor()’:
/home/XXX/progs/OpenCPN/plugins/chartdldr_pi/src/chartdldr_pi.cpp:219:12: error: ‘PLUGIN_VERSION_MAJOR’ was not declared in this scope
     return PLUGIN_VERSION_MAJOR;
            ^
Error #2:
/home/XXX/progs/OpenCPN/plugins/chartdldr_pi/src/chartdldr_pi.h:53:25: fatal error: ocpn_plugin.h: No such file or directory

nohal, 2016-01-05:
Works for me exactly as it should on the Ubuntu 15.10 (and all the packages on Launchpad built successfully just a few commits ago, so it is very unlikely that the build is broken).
Aren't you trying to build just the plugin instead of the whole OpenCPN tree in which it is included?
Ignore the out-of-tree build instructions from the original repository, they are irrelevant since the plugin was merged into the OpenCPN tree.
If you won't resolve it, attach the complete cmake and make output.

jevonlongdell, 2016-01-05:
Thanks for your help. The problem went away when I made a fresh clone of the repository. 

Not sure how I managed to get a non-functioning working copy according to git it was all synced up with this one. Not sure that it is worth looking into further but I attach the cmake output for the original (somehow broken) working copy.

Thanks for your help and sorry about the red herring.

-- *** Build Architecture is amd64
-- *** Staging to build opencpn 4.1.1301 ***
-- *** Package will NOT include documentation ***
-- *** Package will NOT include GSHHS data ***
-- *** Package will NOT include tide and current data ***
-- Found OpenGL....
--     GL Lib: /usr/lib/x86_64-linux-gnu/libGLU.so/usr/lib/x86_64-linux-gnu/libGL.so
--     GL Include: /usr/include
--     Adding local GLU
--     Revised GL Lib: GLU_static/usr/lib/x86_64-linux-gnu/libGL.so
-- Found wxWidgets...
--  wxWidgets Libraries: -L/usr/lib/x86_64-linux-gnu;-pthread;;;-lwx_gtk2u_gl-3.0;-lwx_baseu_net-3.0;-lwx_baseu_xml-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_aui-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu-3.0
--  Revised wxWidgets Libraries: -L/usr/lib/x86_64-linux-gnu;-pthread;-lwx_gtk2u_gl-3.0;-lwx_baseu_net-3.0;-lwx_baseu_xml-3.0;-lwx_gtk2u_html-3.0;-lwx_gtk2u_adv-3.0;-lwx_gtk2u_aui-3.0;-lwx_gtk2u_core-3.0;-lwx_baseu-3.0
-- 
-- *** Staging to build opencpn  ***
-- Build type: Release
-- *** Will install to /usr/local  ***
-- Crash reporting enabled
-- Using Linux crash reporting
-- Found Portaudio: /usr/lib/x86_64-linux-gnu/libportaudio.so
-- Portaudio Found
-- wxCurl configured...
-- SVG support enabled...
-- Building against GTK2...
-- Building with tinyxml includes
-- S57 ENC support: enabled
-- Garmin Host Mode support: enabled
-- Compiling texture compression library with sse support
-- 
-- *** Staging to build PlugIns ***
-- 
-- *** Staging to build dashboard_pi ***
-- *** Staging to build grib_pi ***
-- *** Staging to build chartdldr_pi ***
-- Found OpenGL...
--     Lib: /usr/lib/x86_64-linux-gnu/libGLU.so/usr/lib/x86_64-linux-gnu/libGL.soGLU_static/usr/lib/x86_64-linux-gnu/libGL.so
--     Include: /usr/include
-- Chartdldr_pi building with system tinyxml
-- *** Staging to build wmm_pi ***
-- Found OpenGL...
--     Lib: /usr/lib/x86_64-linux-gnu/libGLU.so/usr/lib/x86_64-linux-gnu/libGL.soGLU_static/usr/lib/x86_64-linux-gnu/libGL.so
--     Include: /usr/include
-- Found OpenGL...
--     Lib: /usr/lib/x86_64-linux-gnu/libGLU.so/usr/lib/x86_64-linux-gnu/libGL.so/usr/lib/x86_64-linux-gnu/libGLU.so/usr/lib/x86_64-linux-gnu/libGL.soGLU_static/usr/lib/x86_64-linux-gnu/libGL.so
--     Include: /usr/include
-- Writing spec file...
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jevon/progs/OpenCPN/build

closed 2016-01-05 by jevonlongdell