OpenCPN/OpenCPN #598: Issue with internationalisation

jongough, 2016-02-10:
There are multiple assert errors under windows 7 in debug mode when running OCPN 4.2. It would appear these are caused by setting the wxLocale and then using setlocale to change the numeric format. Further, trying to use international support within plugins requires timing sensitve changes to the locale which affect the whole application.

OCPN appears to be running with LC_NUMERIC set to "C", which is the default US English setting. This is fine for making exportable files for transfer of information, but internally it makes it very difficult to represent numbers on dialogs, popups and rollovers correctly for the locale of the user.
The following is an example error message.

[image]

The current implementation of locale is not easy to use from a plugin and is subject to random (time sensitive) errors.

Jon

bdbcat, 2016-02-10:
jon...

I see.  Not so easy, actually.  Consider Vadim's comments here:

http://compgroups.net/comp.soft-sys.wxwindows/fine-grained-control-over-wxlocale/2685923

We would need to change all instances of reading/writing decimal numbers at the exact time of access to use "." always.  This means we can't use most of wxWidgets built in functions for numeric conversion, such as wxString::ToDouble().
And we must stop using setlocale(LC_NUMERIC) at all, so letting the global wxLocale setting prevail.

It gets complicated:  For instance, we must use "." in the config file, unless we are willing to wipe the config file and start over with blank file on a user language change.  And XML file i/o must be "." independent of locale, etc, etc...

But I see your point, so...
On the list for O5, I guess.

Dave

nohal, 2019-03-12:
A lot has changed with wx3.1 and the current OpenCPN code, if no objections appear, this issue will be closed on next sweep.

closed 2021-04-11 by jongough