OpenCPN/OpenCPN #654: Android build fails on missing wxQtGesture.h

davemetocean, 2016-05-29:
/OpenCPN/include/glChartCanvas.h:38:39: fatal error: wx/qt/private/wxQtGesture.h: No such file or directory

Weird thing is wxQtGesture.h nowhere to be found in the wxWidgets library. In fact a google search only references the OpenCPN source files. Is this some kind of custom version of wxWidgets?

Any insight appreciated.

nohal, 2016-05-29:
https://github.com/bdbcat/wxWidgets/blob/master/include/wx/qt/private/wxQtGesture.h

davemetocean, 2016-06-11:
That bit works great - thanks. One more question if anyone can help before I close the issue...

Next problem is that we appear to be missing some includes for GLES. From what I can figure out we only have GLES2 headers included. I am building with Qt5.5. Do I need to build and install glshim separately?

Have read the buildandroid/README, but have to say not clear at all what the correct build procedure is.

Build traceback below:

/home/dave/opencpn/OpenCPN/src/glChartCanvas.cpp: In function 'void vertexCallbackD(GLvoid*)':
/home/dave/opencpn/OpenCPN/src/glChartCanvas.cpp:2746:36: error: 'glVertex3dv' was not declared in this scope
     glVertex3dv( (GLdouble *)vertex);
                                    ^
/home/dave/opencpn/OpenCPN/src/glChartCanvas.cpp: In member function 'void glChartCanvas::Render()':
/home/dave/opencpn/OpenCPN/src/glChartCanvas.cpp:4074:19: error: 'GL_PROJECTION' was not declared in this scope
     glMatrixMode (GL_PROJECTION);
                   ^
/home/dave/opencpn/OpenCPN/src/glChartCanvas.cpp:4078:18: error: 'GL_MODELVIEW' was not declared in this scope
     glMatrixMode(GL_MODELVIEW);
                  ^
/home/dave/opencpn/OpenCPN/src/glChartCanvas.cpp:4311:91: error: 'gluPerspective' was not declared in this scope
         gluPerspective(2*180/PI*atan2((double)h, (double)w), (GLfloat) w/(GLfloat) h, 1, w);
                                                                                           ^
/home/dave/opencpn/OpenCPN/src/glChartCanvas.cpp:4321:22: error: 'GL_PROJECTION_MATRIX' was not declared in this scope
         glGetDoublev(GL_PROJECTION_MATRIX, projmatrix);
                      ^
/home/dave/opencpn/OpenCPN/src/glChartCanvas.cpp:4321:54: error: 'glGetDoublev' was not declared in this scope
         glGetDoublev(GL_PROJECTION_MATRIX, projmatrix);
                                                      ^
/home/dave/opencpn/OpenCPN/src/glChartCanvas.cpp:4322:22: error: 'GL_MODELVIEW_MATRIX' was not declared in this scope
         glGetDoublev(GL_MODELVIEW_MATRIX, mvmatrix);
                      ^
CMakeFiles/gorp.dir/build.make:2153: recipe for target 'CMakeFiles/gorp.dir/src/glChartCanvas.cpp.obj' failed
make[2]: *** [CMakeFiles/gorp.dir/src/glChartCanvas.cpp.obj] Error 1

bdbcat, 2021-04-13:
Corrected with current codebase

closed 2021-04-13 by bdbcat