OpenCPN/OpenCPN #676: linker fails on musl libc

ScrumpyJack, 2016-07-04:
While building opencpn 4.2 from github source on musl libc seems to be ok, the last linking bit fails with the following error.
Soooo close...
Let me know if you need more info

> [100%] Linking CXX executable opencpn
/usr/bin/cmake -E cmake_link_script CMakeFiles/opencpn.dir/link.txt --verbose=1
/usr/bin/c++   -Os -fomit-frame-pointer  -pthread -O3 -DNDEBUG   -Wl,--as-needed CMakeFiles/opencpn.dir/src/chart1.cpp.o CMakeFiles/opencpn.dir/src/bbox.cpp.o CMakeFiles/opencpn.dir/src/ocpn_pixel.cpp.o CMakeFiles/opencpn.dir/src/ocpndc.cpp.o CMakeFiles/opencpn.dir/src/chartdb.cpp.o CMakeFiles/opencpn.dir/src/chartdbs.cpp.o CMakeFiles/opencpn.dir/src/chartimg.cpp.o CMakeFiles/opencpn.dir/src/chcanv.cpp.o CMakeFiles/opencpn.dir/src/ocpCursor.cpp.o CMakeFiles/opencpn.dir/src/TCWin.cpp.o CMakeFiles/opencpn.dir/src/S57QueryDialog.cpp.o CMakeFiles/opencpn.dir/src/GoToPositionDialog.cpp.o CMakeFiles/opencpn.dir/src/CM93DSlide.cpp.o CMakeFiles/opencpn.dir/src/RolloverWin.cpp.o CMakeFiles/opencpn.dir/src/ChInfoWin.cpp.o CMakeFiles/opencpn.dir/src/AISTargetQueryDialog.cpp.o CMakeFiles/opencpn.dir/src/AIS_Bitstring.cpp.o CMakeFiles/opencpn.dir/src/AISTargetListDialog.cpp.o CMakeFiles/opencpn.dir/src/AISTargetAlertDialog.cpp.o CMakeFiles/opencpn.dir/src/AIS_Decoder.cpp.o CMakeFiles/opencpn.dir/src/AIS_Target_Data.cpp.o CMakeFiles/opencpn.dir/src/OCPNListCtrl.cpp.o CMakeFiles/opencpn.dir/src/Quilt.cpp.o CMakeFiles/opencpn.dir/src/Hyperlink.cpp.o CMakeFiles/opencpn.dir/src/NavObjectCollection.cpp.o CMakeFiles/opencpn.dir/src/RoutePoint.cpp.o CMakeFiles/opencpn.dir/src/Route.cpp.o CMakeFiles/opencpn.dir/src/SelectItem.cpp.o CMakeFiles/opencpn.dir/src/Select.cpp.o CMakeFiles/opencpn.dir/src/FontMgr.cpp.o CMakeFiles/opencpn.dir/src/FontDesc.cpp.o CMakeFiles/opencpn.dir/src/OCPNSoundData.cpp.o CMakeFiles/opencpn.dir/src/OCPN_Sound.cpp.o CMakeFiles/opencpn.dir/src/NMEALogWindow.cpp.o CMakeFiles/opencpn.dir/src/TTYWindow.cpp.o CMakeFiles/opencpn.dir/src/TTYScroll.cpp.o CMakeFiles/opencpn.dir/src/SendToGpsDlg.cpp.o CMakeFiles/opencpn.dir/src/Layer.cpp.o CMakeFiles/opencpn.dir/src/concanv.cpp.o CMakeFiles/opencpn.dir/src/cutil.cpp.o CMakeFiles/opencpn.dir/src/georef.cpp.o CMakeFiles/opencpn.dir/src/navutil.cpp.o CMakeFiles/opencpn.dir/src/routeman.cpp.o CMakeFiles/opencpn.dir/src/routemanagerdialog.cpp.o CMakeFiles/opencpn.dir/src/routeprop.cpp.o CMakeFiles/opencpn.dir/src/routeprintout.cpp.o CMakeFiles/opencpn.dir/src/PositionParser.cpp.o CMakeFiles/opencpn.dir/src/printtable.cpp.o CMakeFiles/opencpn.dir/src/chartbarwin.cpp.o CMakeFiles/opencpn.dir/src/tcmgr.cpp.o CMakeFiles/opencpn.dir/src/Station_Data.cpp.o CMakeFiles/opencpn.dir/src/IDX_entry.cpp.o CMakeFiles/opencpn.dir/src/TCDataFactory.cpp.o CMakeFiles/opencpn.dir/src/TCDS_Ascii_Harmonic.cpp.o CMakeFiles/opencpn.dir/src/TCDS_Binary_Harmonic.cpp.o CMakeFiles/opencpn.dir/src/TCDataSource.cpp.o CMakeFiles/opencpn.dir/src/thumbwin.cpp.o CMakeFiles/opencpn.dir/src/options.cpp.o CMakeFiles/opencpn.dir/src/gshhs.cpp.o CMakeFiles/opencpn.dir/src/kml.cpp.o CMakeFiles/opencpn.dir/src/undo.cpp.o CMakeFiles/opencpn.dir/src/about.cpp.o CMakeFiles/opencpn.dir/src/ais.cpp.o CMakeFiles/opencpn.dir/src/pluginmanager.cpp.o CMakeFiles/opencpn.dir/src/gpxdocument.cpp.o CMakeFiles/opencpn.dir/src/wxJSON/jsonwriter.cpp.o CMakeFiles/opencpn.dir/src/wxJSON/jsonreader.cpp.o CMakeFiles/opencpn.dir/src/wxJSON/jsonval.cpp.o CMakeFiles/opencpn.dir/src/styles.cpp.o CMakeFiles/opencpn.dir/src/toolbar.cpp.o CMakeFiles/opencpn.dir/src/compass.cpp.o CMakeFiles/opencpn.dir/src/geodesic.cpp.o CMakeFiles/opencpn.dir/src/datastream.cpp.o CMakeFiles/opencpn.dir/src/OCPN_DataStreamEvent.cpp.o CMakeFiles/opencpn.dir/src/ConnectionParams.cpp.o CMakeFiles/opencpn.dir/src/OCP_DataStreamInput_Thread.cpp.o CMakeFiles/opencpn.dir/src/multiplexer.cpp.o CMakeFiles/opencpn.dir/src/pugixml.cpp.o CMakeFiles/opencpn.dir/src/OCPNRegion.cpp.o CMakeFiles/opencpn.dir/src/LLRegion.cpp.o CMakeFiles/opencpn.dir/src/TrackPropDlg.cpp.o CMakeFiles/opencpn.dir/src/LinkPropDlg.cpp.o CMakeFiles/opencpn.dir/src/ssl/sha1.c.o CMakeFiles/opencpn.dir/src/viewport.cpp.o CMakeFiles/opencpn.dir/src/canvasMenu.cpp.o CMakeFiles/opencpn.dir/src/OCPNPlatform.cpp.o CMakeFiles/opencpn.dir/src/crashprint.cpp.o  -o opencpn  -L/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0 -rdynamic libNMEA0183.a -lportaudio libWXCURL.a -lcurl libWXSVG.a libS57ENC.a libGARMINHOST.a libSYMBOLS.a -pthread -Wl,--as-needed -lwx_baseu_net-2.8 -lwx_baseu_xml-2.8 -lwx_gtk2u_html-2.8 -lwx_gtk2u_adv-2.8 -lwx_gtk2u_aui-2.8 -lwx_gtk2u_core-2.8 -lwx_baseu-2.8 -lbz2 -lz -lSM -lICE -lX11 -lXext -ltinyxml -ldl -lcairo -lpangocairo-1.0 -lpangoft2-1.0 -lpangoxft-1.0 -lgdk_pixbuf-2.0 -lexpat -lglib-2.0 -lgobject-2.0 -latk-1.0 -lgio-2.0 -lgthread-2.0 -lgmodule-2.0 -lpango-1.0 -lcairo -lpangocairo-1.0 -lpangoft2-1.0 -lpangoxft-1.0 -lgdk_pixbuf-2.0 -lexpat -lglib-2.0 -lgobject-2.0 -latk-1.0 -lgio-2.0 -lgthread-2.0 -lgmodule-2.0 -lpango-1.0 -lgdk-x11-2.0 -lgtk-x11-2.0 
CMakeFiles/opencpn.dir/src/LLRegion.cpp.o: In function `LLerrorCallback(unsigned int)':
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:310: undefined reference to `gluErrorString'
CMakeFiles/opencpn.dir/src/LLRegion.cpp.o: In function `LLRegion::PutContours(work&, LLRegion const&, bool)':
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:469: undefined reference to `gluTessBeginContour'
CMakeFiles/opencpn.dir/src/LLRegion.cpp.o: In function `work::PutVertex(contour_pt const&)':
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:267: undefined reference to `gluTessVertex'
CMakeFiles/opencpn.dir/src/LLRegion.cpp.o: In function `LLRegion::PutContours(work&, LLRegion const&, bool)':
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:476: undefined reference to `gluTessEndContour'
CMakeFiles/opencpn.dir/src/LLRegion.cpp.o: In function `work::PutVertex(contour_pt const&)':
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:267: undefined reference to `gluTessVertex'
CMakeFiles/opencpn.dir/src/LLRegion.cpp.o: In function `work::work(LLRegion&)':
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:248: undefined reference to `gluNewTess'
CMakeFiles/opencpn.dir/src/LLRegion.cpp.o: In function `LLRegion::Put(LLRegion const&, int, bool)':
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:484: undefined reference to `gluTessCallback'
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:485: undefined reference to `gluTessCallback'
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:486: undefined reference to `gluTessCallback'
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:487: undefined reference to `gluTessCallback'
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:488: undefined reference to `gluTessCallback'
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:489: undefined reference to `gluTessProperty'
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:490: undefined reference to `gluTessProperty'
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:493: undefined reference to `gluTessNormal'
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:495: undefined reference to `gluTessBeginPolygon'
CMakeFiles/opencpn.dir/src/LLRegion.cpp.o: In function `LLRegion::PutContours(work&, LLRegion const&, bool)':
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:469: undefined reference to `gluTessBeginContour'
CMakeFiles/opencpn.dir/src/LLRegion.cpp.o: In function `work::PutVertex(contour_pt const&)':
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:267: undefined reference to `gluTessVertex'
CMakeFiles/opencpn.dir/src/LLRegion.cpp.o: In function `LLRegion::PutContours(work&, LLRegion const&, bool)':
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:476: undefined reference to `gluTessEndContour'
CMakeFiles/opencpn.dir/src/LLRegion.cpp.o: In function `LLRegion::Put(LLRegion const&, int, bool)':
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/LLRegion.cpp:500: undefined reference to `gluTessEndPolygon'
CMakeFiles/opencpn.dir/src/LLRegion.cpp.o: In function `work::~work()':
[... cut ...]
/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0/src/s52plib.cpp:1111: undefined reference to `glDeleteLists'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/opencpn.dir/build.make:2377: opencpn] Error 1
make[2]: Leaving directory '/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0'
make[1]: *** [CMakeFiles/Makefile2:363: CMakeFiles/opencpn.dir/all] Error 2
make[1]: Leaving directory '/home/scrumpyjack/aports/testing/opencpn/src/OpenCPN-4.2.0'
make: *** [Makefile:153: all] Error 2

nohal, 2016-07-04:
Not that it has anything to do with your problem, but is there any real reason to use wx2.8 on your platform? And is there any reason not to use the current OpenCPN codebase?
Most errors above seem that you are not linking against GLU library, can we see the ```cmake ..``` output as well?
The missing ```backtrace_symbols``` bit can be worked around setting ```OCPN_USE_CRASHREPORT=OFF``` during the configuration step, won't have any impact on usability.

nohal, 2016-07-04:
I suppose you are trying to build on Alpine Linux:

1. It seems to me that the WX2.8 package on Alpine is built without OpenGL support. One suitable for OpenCPN should probably look something like https://github.com/nohal/aports/commit/bc847daa7c75903bc6055a51a88545f4da0d6751 (ATM I am definitely not going to become an Alpine Linux contributor though, so if it has to be submitted upstream, someone else has to take the responsibility)
2. I don't see any problem with the GLU stuff when ```glu-dev``` (or something else I installed while playing with the above) is installed. 
3. Musl does not have ```backtrace*``` support, but there is ```libexecinfo``` which provides the needed stuff - simply install ```libexecinfo-dev``` and patch OpenCPN's CMakeLists.txt to link against it with something like 
```
index 967e558..27bad69 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -489,6 +489,9 @@ IF(OCPN_USE_CRASHREPORT)
             IF(CMAKE_BUILD_TYPE MATCHES "Rel*")
                    MESSAGE (STATUS "Using Linux crash reporting")
                 ADD_DEFINITIONS(-DLINUX_CRASHRPT)
+                IF(EXISTS /etc/alpine-release)
+                    SET(EXTRA_LIBS ${EXTRA_LIBS} "execinfo")
+                ENDIF(EXISTS alpine-release)
             ENDIF(CMAKE_BUILD_TYPE MATCHES "Rel*")
         ENDIF(NOT APPLE AND NOT QT_ANDROID AND NOT MINGW)
     ENDIF(MSVC)
```
(I'm open to including this patch into OpenCPN's master once there really is an Alpine Linux packaging effort)