OpenCPN/OpenCPN #686: NCPUCount no longer works

NAHANNIV, 2016-07-23:
NCPUCount was introduced to allow the use of all processor cores during texture cache rebuild on platforms where CPU cores are dynamically enabled/disabled based on workload (specifically Nvidia Jetson).

As of some recent changes this no longer works.

did-g, 2017-04-01:
Hi,
Could test with 4.6.1, it should work again;

Thanks
Didier

NAHANNIV, 2017-04-01:
Thanks, I will test it as soon as I have a good internet connection and some time (we are cruising now).

nohal, 2018-01-15:
@NAHANNIV, did you manage to test? And if not, can you with the current stuff?

nohal, 2019-03-12:
Will leave it around until the next sweep, if no new info appears, will be closed.

FredericGuilbault, 2019-03-28:
I report it working with 5.0.0  If that can help

nohal, 2019-03-29:
Thanks for the info.

closed 2019-03-29 by nohal