OpenCPN/OpenCPN #687: Application can quit on OSX

MartinDelille, 2016-07-27:
I had to kill the application because the quit menu action was greyed:
[image]

I'm using the latest release 4.4.0

nohal, 2016-07-27:
Any steps to reproduce?  Didn't you have some modal dialog (Like the Toolbox) open? In this state your report is unfortunately totally useless.

MartinDelille, 2016-07-27:
No yet but I'll tell you if I manage to reproduce it. Anyway, this is not a very good design to block application exit menu item even if there is a modal window opened (which is not the case here).

MartinDelille, 2016-07-27:
Forget my last comment: it seems that it is a common behavior. Maybe I missed something.

closed 2016-07-27 by MartinDelille

nohal, 2016-07-27:
It simply is the price for being cross-platform - As you sure know, on other platforms the application menu is part of the application window = not even accessible when a modal dialog is open...

MartinDelille, 2016-07-27:
It's fine. I have a big multiplateforme too: https://github.com/phonations/joker

For my information, do you use Qt? Which version?

nohal, 2016-07-27:
No, we do not use Qt on the desktop platforms, it is the backend under wxWidgets in the Android port.