OpenCPN/OpenCPN #698: Support for `USE_GL=OFF` is not maintained anymore

SamsonovAnton, 2016-08-07:
As ticket 1841 was removed from Flyspray tracker, I am asking this once again here.

Developer's guide, particularly “[Compiling — Linux](http://opencpn.org/ocpn/compiling_source_linux)”, states:

> From 3.3.117 is it possible to build without using OpenGL at all.

However, that seems no longer true in practice, at least for me. Even further, more and more pieces of code are updated without `#ifdef ocpnUSE_GL` in mind; dumb fixes like in #619 are far from enough now.

Should not it be stated then that OpenCPN absolutely needs OpenGL, and that `USE_GL` must be `ON` at all times? May be it is time to remove `USE_GL` from configurable values and set it to `ON` implicitly?

nohal, 2016-08-07:
Builds and works for me. Any errors in particular you see? That without OpenGL more and more functionality will simply be unavailable is true and there is nothing we can do about it due to performance reasons.

I wonder how FS#1841 could have been deleted - not even I can do it with admin privileges there...

SamsonovAnton, 2016-08-07:
Are you sure?

The first error I get is:

> …/src/chcanv.cpp:10166:45: error: ‘class RoutePoint’ has no member named ‘m_wpBBox’
>    if( !BltBBox.IntersectOut( pWP->m_wpBBox ) )

Indeed, `m_wpBBox` is only defined inside `#ifdef ocpnUSE_GL`, so it is not visible here. But if I put the very same `#ifdef ocpnUSE_GL` around this particular line (as a temporary solution), more errors pop up at linking stage, because *piano.cpp* has now featuring multiple unconditional references to `glColor4ub`, `glTexImage2D`, etc., — which cannot be simply put into conditional block without any other alternative code path.

nohal, 2016-08-07:
Ah, I wasn't at the very tip of the master. This fixes the build there for me:
[fix_nogl.diff.txt](https://github.com/OpenCPN/OpenCPN/files/405471/fix_nogl.diff.txt)

nohal, 2016-08-07:
Or do you do something more to get the non-GL build than
`cmake -DUSE_GL=OFF && make`?

SamsonovAnton, 2016-08-07:
Not doing any extra actions, just trying to build from *master* with `USE_GL=OFF`. After applying your patch, I still get *undefined external reference* error with regard to `glColor4ub` while trying to link in *piano.cpp.o* (your patch does not modify that file). Are you sure OpenGL libraries are not linked automatically in your build configuration?

FYI. OpenGL-enabled builds are created just fine in my setup.

nohal, 2016-08-07:
No, I'm not sure that GL libs are not linked in automatically in my particular config, it is well possible. I just don't think that asking to remove the feature as it is currently broken in the master branch which is not even in beta is a bit heavy. Taking it a bit further, we should also remove support for Mac which is currently also broken...

SamsonovAnton, 2016-08-07:
I am not asking to remove anything (why would I?), — just trying to figure out what is the official attitude towards that matter, taking into account that some legacy features may be indeed declared deprecated in the next major release and that this topic particularly was already raised in the past with no definitive assurance about its future.

Quite the opposite, I am personally voting to keep non-GL builds possible as long as it seems practical. For example, I mostly experiment with OpenCPN on a machine with an onboard Intel graphics, which exhibits inferior performance in GL mode. Moreover, I often run my Linux system inside a VirtualBox machine, which either isolates guest from the hardware or adds just another layer of abstraction, not only breaking performance even further, but also introducing glitches and critical bugs that lead to guest and/or host crashes. However, I do not nearly think that such setup is representative for a regular OpenCPN user.

rgleason, 2016-08-07:
Anton, I've searched for #1841 and I think I recall it about the use of gl. It does not exist in the closed status and is not an open tracker item.
Perhaps it was damaged during one of the tracker database crashes? I know that tracker has had to be restarted several times and on one occasion even changed the url.

nohal, 2016-08-07:
Interesting. 100% of my Windows and Linux work on OpenCPN is done in VirtualBox VMs, with OpenGL on all the time. Never seen a crash. For a long time on a machine with onboard Intel GPU. GL performance always better than non-GL. I must be very lucky seeing the hell you must live in :)

SamsonovAnton, 2016-08-07:
Pavel, that is not anywhere near hell — it is just a slight inconvenience. The real hell looks more like 500 MHz SPARC32 CPU with custom 2D-only video controller without acceleration capabilities (where each screen update takes up to several seconds), with Linux 2.4.25, LibC 2.5, GCC 3.3 and other software of that era. Surprisingly, OpenCPN can be made running in such harsh environment (or at least it was a year ago), — despite that it is purposeless, since O is (was) totally unaware of byte order other than little-endian and restrictions posed by aligned-only memory access.

nohal, 2016-08-07:
This has to end soon, every cool project now must use C++11, or better C++14 and not even build with anything older than the latest CLang and VS2015 ;)

bdbcat, 2016-08-07:
There is no set plan to require GL on O5.  I think this would be a reversion for many valid use cases, especially common older intel platforms in general use, and even new W10 updates with broken intel drivers.  sigh....

So, should make at least a decent effort to build and run O5 without GL.  If this means that we need to revert/modify some of the commits on Master, so be it.

So, lets dig into it, taking compile/link errors one by one.  I'll start when I get a new air tank....

Dave

nohal, 2019-03-12:
Still not buildable with `-DUSE_GL=OFF`, moving to next milestone

caesar, 2021-12-19:
In the course of trying to do an arm64 build on macOS (#2486), I have managed to build with `-DOCPN_USE_GL=OFF` by adding a *ton* of `ifdef`s.

Assuming there is still interest in keeping OCPN buildable without OpenGL support (I'd argue it's good to have that option working if possible), I will submit a PR after I have done some testing (not today…)

bdbcat, 2021-12-19:
I am interested.  I don't know anyone using this option today.  But what I have learned is that "you can never remove legacy code".  Somebody, somewhere, will miss it.  And for something this pervasive in the code, if it goes away, it will be gone for good.
Thanks
Dave
btw, Welcome back to the team.  Long time no see....

caesar, 2021-12-20:
Great stuff Dave, in that case once I have everything working properly on my M1/Arm64 Mac, I'll do some proper testing to make sure I didn't break anything, and submit a PR. As you say, probably not many people use it, but if it's easy to make it work that's better than removing it at least for now.

> btw, Welcome back to the team. Long time no see....

Thanks! I've been super busy but always meaning to get back into contributing, especially to fix a few pet peeves of mine ;-) I have quite a list of features I'd like to see improved or added, so hopefully you'll be seeing me around more again.

nohal, 2024-02-15:
@bdbcat what do we want to do here?
For the fun of it I tried to build with  `-DOCPN_USE_GL=OFF` and with a couple of `ifdefs` it gets to linking where `s52plib` is missing `glew`.

We have basically two options, throw in the towel and start to remove this thing completely with all the `ocpnUSE_GL` `ifdefs` or try to get it to build something at `DisableOpenGL=1` level, start doing CI builds for this configuration and require all new code to pass.

bdbcat, 2024-02-15:
I'll take this one.
I'd like to retain no-GL mode functionality.  Should not be so hard to do.  All the code is there, just not completely tied together.  I agree that we must integrate to CI, else we will just lose it again.

nohal, 2024-02-27:
I have fixed the few little issues left and enabled the CI build, so further breakage will be spotted as it happens.

closed 2024-02-27 by nohal

bdbcat, 2024-02-27:
Thanks for the fixes.
Don't know how I missed that.  I rebuilt clean many, many times....