OpenCPN/OpenCPN #709: Opencpn crashes when trying to display ENC chart

CptanPanic, 2016-08-11:
I am trying to load ENC charts from Florida, and when it tries to actually display them the application crashes.  The RNC ones work fine.  This is on ubuntu 16.04.  Any settings I can try?

nohal, 2016-08-11:
No setting that fixes crashes is available.
You can try to have a look into the logfile to find out on which chart the application crashes and in case you manage to isolate it, post it for analysis or otherwise create a crash scenario we could reproduce.
You can try to get fresh charts.
You can try disabling OpenGL if it is enabled or vice versa.
You can run opencpn from command line and see whether it produces a backtrace to help identify the cause of the crash.
You can use gdb or another debugger to get the backtrace and help identify the cause of the crash.
You can try ```mv ~/.opencpn ~/.opencpn.bak``` while OpenCPN is not running and see whether something was rotten in your local OpenCPN data. 

For questions please use the forum at http://www.cruisersforum.com/forums/f134/

FWIW I have just downloaded all the 99 FL ENC charts using the integrated chart downloader and see no crash viewing them in OpenCPN 4.4.

closed 2016-08-11 by nohal

rgleason, 2016-08-11:
Added your response to FAQ
http://opencpn.org/ocpn/faq