OpenCPN/OpenCPN #717: Linux version hides GRIB plugin if you toogle fullscreen (F11)

Klezstyle, 2016-09-14:
This happens with master compiled from git in a Arch Linux 64bit machine with "wxgtk 3.0.2" (stable) installed. OpenCPN reports as version 4.5

I´m using Gnome.

Launch OpenCPN, open the grib window and then go to fullscreen hitting F11. The GRIB window becomes hidden. Hit again F11 to go back to windowed mode and the GRIB window becomes visible again.

Also, in fullscreen mode (F11) it´s impossible to move the toolbar to the uppermost side of the screen, it keeps a little displaced from the top. See attached screenshot.
[image]

nohal, 2018-01-15:
The fullscreen toggle is definitely "broken" in different ways on different window managers and needs another round of the trial and error (On current Unity on Ubuntu 16.04 the launcher and menubar stay on the screen while the native apps like Terminal manage to hide them after F11. The Dashboard Z-Order issue is not present there though and seems to be Gnome's particular "feature")

nohal, 2019-03-12:
The issue seems gone in the current code (on Ubuntu Bionic, Gnome Shell).

closed 2019-03-12 by nohal