OpenCPN/OpenCPN #738: Touchscreen issue open cpm 4.4/4.2 Rpi 2 Rasbian

Garem, 2016-12-08:
Hi folks,

I have an issue with the control buttons:
https://vid.me/TKi7

Setup:
Raspberry Pi 2, Opencpn 4.4 (4.2 aswell) powered by openplplotter image, 5inch HDMI LCD (B) (Firmware Rev 2.1)

tmulzer, 2016-12-08:
I have the same on pi3 with the official 7" capacitive touchscreen.
Raspbian of spring 2016.

nohal, 2018-01-31:
Still reproducible with the fixes in current code?

bdbcat, 2018-02-04:
I have the hardware to test.
However, the video link is dead, so no details on the actual bug are available.

nohal, 2019-03-12:
@Garem, @tmulzer still reproducible? Any chance for a new video or better description of the steps to reproduce?

Hakansv, 2019-03-12:
Pavel.. I have just tested on my RPi with the 7 inch screen. All buttons and dialog tested are easy to check and use. No problem to navigate in options.

closed 2019-03-12 by nohal