OpenCPN/OpenCPN #753: Periodic configuration reset on Rasperry Pi

JonathanHuot, 2017-01-17:
Hi,
I plugged a Raspberry Pi 3 on my boat (HDMI TV is on 12v and Rasperry Pi is on TV's USB). It works very well.

However, periodically, when I shutdown and restart it, the configuration is reset with its defaults (prompring the agreement message).
So I'm forced to redo all configurations (adding charts, creating groups, set lang, and so on...).

Issue has appeared twice in a week, so it's very annoying.
I'm trying to reproduce it constantly but I didn't found how to do so.

I've checked the opencpn code, but SetDefaultOptions looks safe to me. Have you any ideas ?

rgleason, 2017-01-17:
Jonathon This is not really the place for this question. I think you should post it on Cruisers Forum opencpn in an rpi thread.  However you know what the problem is most likely to be.... power! Get a good separate supply.
You can close this now and post in CF these kinds of questions.

JonathanHuot, 2017-01-17:
Hi, Thanks for your prompt reply. I will open a question on CF and probably test with a separate power supply.

However, how could a power issue be related to a configuration reset ? Even when a power outage, opencpn should not reset/loose the configuration files, am I right ? Does OpenCPN write periodically to this file ? If yes, then could it be wxWidgets which is not "atomically" writing to the filesystem ? If again yes, a race condition can happen if both wxWidgets write to disk and a power outage come? How couldn't we trust wxWidgets?

Thanks again and sorry to dig more in this topic.

nohal, 2017-01-17:
How do you "shutdown and restart" your Pi? Cutting the power?

JonathanHuot, 2017-01-17:
Yes

nohal, 2017-01-17:
Then it is not very surprising that the filesystem gets corrupted, is it? `shutdown -h` should send the `SIGTERM` signal correctly to all the running applications.

rgleason, 2017-01-17:
Ah the power of the user... never would have thought it would be treated like an appliance.

JonathanHuot, 2017-01-17:
I understand than FS corruption can happen from time to time. Even without being installed on a Rasperry Pi, it can happen. 

However, since the main aim of this fabulous software is to be on a boat sailing the seas, where power and all devices onboard can't be reliable (don't you have spare for everything?). I'm still hoping we could improve the reliability of the *configuration*.

Improving reliability of the overral configuration means doing less write to disk. For example, I found that last known boat's coordinates are inside the same configuration file as the program settings are.
Let's say, if we store all dynamic informations into a separate file and all static configuration into another, even if we have a FS corrupted file, guesses are that it will no longer reset your global settings. Most importantly when sailing, you probably don't have time (or knowledge) to reconfigure everything.

What do you think ? I'm a developer and not an user, so I can help.

rgleason, 2017-01-17:
Jonathon, Sorry. Yes please help. You are over my head here. I said enough. Hope you'll continue to pitch in!

nohal, 2017-01-17:
Yes, the configuration settings can be protected better that they are now by stripping the "dynamic" values remembered there to a separate location and making sure the main config file is open for the least needed amount of time for sure. Or further by implementing some CoW logic around the saving of the config. I don't see any reason why a pull request implementing it should be refused.

nohal, 2018-01-15:
OK, I have found a scenario when this actually happens. After new version installation, in case no settings were changed during the session AND the shutdown is an unclean kill.

commit d266644e (2018-01-15): Persist the answer to the 'not for navigation' warning mesage early in the app init process to prevent loss of it in case of later crash without changing any settings in the session. Fixes #753
closed 2018-01-15 by nohal