OpenCPN/OpenCPN #755: Inclusion in f-droid

robinpaulson, 2017-01-19:
I have asked the f-droid package maintainers to include opencpn in their repositories. See this link to follow progress and answer any queries:
https://gitlab.com/fdroid/rfp/issues/8

See these links for more on F-Droid:
https://f-droid.org/wiki/page/Main_Page
https://en.wikipedia.org/wiki/F-Droid

robinpaulson, 2017-03-13:
I have discussed the inclusion of OpenCPN in f-droid; the developers have remarked on the inclusion of Google Play Services, which of course is non-free. Could someone tell me what function this provides to the software, we may investigate including it without that dependency?

UnderSampled, 2021-02-20:
I'd be willing to provide a bounty for someone to solve this.

encrypt94, 2026-10-08:
If the OpenCPN team is okay with having the app on F-Droid, I'd like to work on this!

encrypt94, 2026-10-29:
any thoughts on this @nohal @bdbcat @leamas ?

rgleason, 2026-10-29:
It is opensource.

bdbcat, 2026-10-30:
In general, I have no objection to making OpenCPN for Android available through F-Droid.  There are some real advantages to F-Droid, not the least of which is potentially increased exposure to the OpenCPN ecosystem.
But there are some issues to consider.  There are four primary points on my mind.

1.  Strictly technical, related to scheduling:
OpenCPN/Android is currently receiving a long overdue refactor of the JNI interfaces between OpenCPN and the Java runtime environment of Android.  The work is ongoing, with come very preliminary alpha testing underway.  All of the source code available for OpenPCN/Android currently published on github should really be considered to be obsolete.  New sources are partially published now, will be complete when the re-factor is released on Playstore.

2.  Technical support:
Technical support of OpenCPN/Android is currently provided by an online forum, which is separate from the generally used CruiserForum.  This arrangement provides increased focus on Android-specific issues arising from the significantly different UX on the Android vs. a traditional desktop environment.  The OpenCPN/Android forum is meant to support users acquiring OpenCPN from the Google Playstore.  However, privacy limitations and GPL license terms preclude the forum operators from enforcing this policy in any meaningful way.  Thus, the potential for excessively high traffic on the forum exists, to the detriment of existing OpenPCN/Android users.

3.  Security and OpenCPN reputation:
F-Droid advertises a strong security model, and there is no obvious reason to mistrust F-Droid's machinery.The idea of building the APK from open source repo is compelling.
And yet... the Google Playstore app distribution model is proven over many yeears to be secure and convenient for the typical Android user.
My concern is solely the reputation of OpenCPN in general, and OpenCPN/Android specifically.  All it takes is one accidental or intentional upload of a malicious source file to tarnish that reputation, perhaps irretrievably.
Using the Playstore interface, we (actually I) own that responsibility, and take it seriously.  Who will responsibly own the F-Droid interface for OpenCPN/Android?

4. Version coherence:
We could certainly expect the case that the F-Droid version and the Playstore version will diverge over time, causing user confusion.  "Which one is the real OpenCPN/Android?"

tl;dr;
At the moment, I would prefer that OpenCPN/Android **_not_** be offered on F-Droid, before some significant points are resolved.