OpenCPN/OpenCPN #757: No license found in code nor in root directory.

JonathanHuot, 2017-01-23:
A "LICENSE" file should be present at root directory including the terms of GPLv2.

nohal, 2018-01-15:
The license is mentioned in the `README` and present in `data/license.txt`, but yes, this stuff needs a lot of cleanup to get us closer to what packagers like Debian require.

leamas, 2018-07-01:
Actually, the license issue is more complicated. Although the official statement is that opencpn is distributed under GPLv2 a simple check reveals at least:

  - BSD/MIT files
  - GPLv3+ files
  - LGPLv2+ and LGPLv3+ files
  - Bundled glu sources using  SGI Free Software License B (v2.0) (MIT derivative?)

All these licenses seems to be GPL-compatible. That said, distributing code built from GPLv3+ sources under GPLv2 is a clear license violation, and should basically block opencpn from being distributed by any distro sensitive to legal threats (which, luckily, Debian isn't).

I don't know how to fix this.  Parts of it could be handled by clarifying the different licenses used in data/copyright or so. The basic fix would be to re-license opencpn to GPLv2+ and distribute it as GPLv3+, but that would mean talking to all of those having copyrighted GPLv2 code in the project. 

Sorry for bringing this rotten fish to the table.

nohal, 2018-07-01:
Have you seen any GPLv3 other than the cmake files in chart downloader plugin + Curl.cmake (Which happen to be mostly mine and I really don't care about the license much)? If it is just those few files, it sure is simpler to relicense those than the whole OpenCPN codebase...

leamas, 2018-07-05:
Attaching a distorted run of licensecheck
[licenses.log.gz](https://github.com/OpenCPN/OpenCPN/files/2166782/licenses.log.gz).  Note that  distributing GPLv2+ files as GPLV2 is also a violation.

nohal, 2018-07-05:
Could you please explain, or point me to a source explaining, how our own "GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or  (at your option) any later version." violates our own GPLv2?

leamas, 2018-07-05:
It certainly doesn't. But there are other variants, like the wording in license.txt which is  *This program is distributed under the terms of the GPL v2* . 

The easy path would be to rectify this and other references to plain GPLv2 to GPLv2+ if it's a reasonable assumption  that's the intended meaning? Which it probably is, since the files indeed are GPLv2+, not GPLv2?

leamas, 2018-07-05:
Looking at the list (still not thoroughly) I find GPLv3+ and also a single Apache-2.0 file (+ some in the android bundle). These are not compatible with GPLv2+, but are fine using GPLv3+.

So., if the basic license (most "own" files) could be clarified to GPLv2+, a simple solution would be to change the overall license to GPLv3+, which looks fine with the files I see. We also need simple copyright file mentioning the different licenses used, but nothing as detailed as the current data/copyright file. That level is something needed only on Debian, and is probably best handled by downstream packagers.    A statement mentioning GPLv2+, GPLV3+, Apache-2.0, SGI  and BSD should be fine

nohal, 2018-07-05:
Well, for my own work I will certainly make sure it never even gets near GPL in the future, that is for sure...
The intended meaning is "When Dave started to work on the code, there was no GPLv3 so he has written GPLv2 into the copyright file". If GPL really is not compatible even with itself, but at the same time allows to switch from older version to newer without any problem (That is what I understand you say, correct me if I'm again wrong), then let's write GPLv3+ wherever there is now GPLv[2,3]+?, be done with it and actually do something useful. Of course if that "But you must get a permission from whoever touched that codebase at any past moment" crap applies even to that, I would gladly let Debian not package OpenCPN, ever.

leamas, 2018-07-05:
Yes... this is kind of boring, agreed.
> then let's write GPLv3+ wherever there is now GPLv[2,3]+?

There is no reason at all to change any file which is GPLv2+, its fine as is it is. It can be distributed under GPLv3+ as well.

>  If GPL really is not compatible even with itself

It's really simple: A GPLv3 license says basically gpl version 3, or any later version. And this does not cover GPLv2, obviously.

GPLv2 is GPLv2, full stop. It can only be distributed as GPLv2. GPLv2+ on the other hand, can be be distributed as GPLv3+.  

>  "When Dave started to work on the code, there was no GPLv3 so he has written GPLv2 into the copyright file".

Hm... have you any contact with him?

leamas, 2018-07-05:
I'm clear as mud, sorry. What I see is a possibility of a copyright file basically stating

   - Distributed under GPLv3+ and GPLv2+ and Apache-2.0+ and BSD/X11.
   - Most files are GPLv2+
   - Some files are GPLv3+, namely...
   . Some files are Apache 2.0, namely...
   - Some files are BSD, namely.

All this is perfectly sane. Furthermore, the core truth comes from the copyrights in the files. And interestingly, I cannot see any GPLv2 there, just GPLv2+. So whatever the history is, I think a copyright according to above could work.

The corresponding license files should be in the root dir as well. I don't really have the time now, but if we could clear the GPLv2+ (no GPLv2) assumption, I could try to make a PR later; I have some tooling at hand for this.

That said: IANAL ...

nohal, 2018-07-05:
@bdbcat - Dave, do you have anything against what's outlined above? If getting rid of any explicit GPLv2 mentions makes this to stop appearing over and over again, we should probably do it...

bdbcat, 2018-07-05:
@pavel, @leamas 

Clear as mud to me also.
I have no objections to a PR that cleans this up in whatever way is simplest.
Dave

nohal, 2019-02-12:
@leamas are we done here? Or at least as close to done as we could handle the rest next time there is enough patience at all sides to try to handle the debian inclusion again?

leamas, 2019-02-12:
The debian package  has passed the sponsor review and is now in the NEW queue [1], the very last step. It's possible that some new things pop up here; let's keep this bug open until we know.  I'll keep it updated.

[1]  https://ftp-master.debian.org/new.html

leamas, 2020-10-17:
The Debian package uploaded, we have passed all reviews. Also: There are multiple license files in the top directory. This bug should IMHO be closed.

@JonathanHuot  @bdbcat : time to close?

JonathanHuot, 2020-10-18:
Yes, very good work and investigation there!

closed 2020-10-18 by JonathanHuot