OpenCPN/OpenCPN #763: Documentation/tutorial to create a plugin?

yohanboniface, 2017-02-04:
I'm certainly missing something, but I can't put my hand on a documentation on how to create a plugin.
I've had a look at https://www.opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:plugins:plugin_api but I can only see generic hight level informations.
Also here https://www.opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:plugins:plugin_documentation but it seems to be "how to write a plugin documentation".
If I search on Google, every link I try ends in 404 (seems broken in a site update).

Any known resource on how to create a plugin?

Thanks!

nohal, 2017-02-04:
Creating a plugin is no rocket science, use the existing ones as an example.
If you want to search the new wiki, use it's integrated search function (the search field is on the top of the page), Google and the other search engines will probably take some more time to update their indexes.

closed 2017-02-04 by nohal

jishatomc, 2019-07-23:
> I'm certainly missing something, but I can't put my hand on a documentation on how to create a plugin.
> I've had a look at [...]

I am new to this OpenCPN, as part of my curriculum project, ineed to create a plug in to draw something on OpenCPN .

Can You help me.

rgleason, 2019-07-23:
Look in the Developer Guide below, do a search using this link, and if you are still having trouble, 
Create a new post on the forum with appropriate title like "Build Plugins" and we will help you. 

https://www.opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide

taruti, 2019-08-02:
Also perhaps https://github.com/OpenCPN/OpenCPN/tree/master/plugins/demo_pi_sample

rgleason, 2019-08-02:
Also Learning
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:plugins:learning