OpenCPN/OpenCPN #773: LNK2005	_XML_ParserCreate already defined in wxexpatd.lib(xmlparse.obj)

pavelpasha, 2017-02-12:
Hi OpenCPN team. While trying to compile on Windows (VS2015) faced with the problem. 
```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1120	1 unresolved externals	chartdldr_pi	C:\Users\virtualBox\Desktop\OpenCPN-4.4.0\build\plugins\chartdldr_pi\Debug\chartdldr_pi.dll	1	
Error	LNK2005	_XML_ParserCreate already defined in wxexpatd.lib(xmlparse.obj)	opencpn	C:\Users\virtualBox\Desktop\OpenCPN-4.4.0\build\expat.lib(expat.dll)	1	
Error	LNK2005	_XML_SetElementHandler already defined in wxexpatd.lib(xmlparse.obj)	opencpn	C:\Users\virtualBox\Desktop\OpenCPN-4.4.0\build\expat.lib(expat.dll)	1	
Error	LNK2005	_XML_SetCharacterDataHandler already defined in wxexpatd.lib(xmlparse.obj)	opencpn	C:\Users\virtualBox\Desktop\OpenCPN-4.4.0\build\expat.lib(expat.dll)	1	
Error	LNK2005	_XML_SetCommentHandler already defined in wxexpatd.lib(xmlparse.obj)	opencpn	C:\Users\virtualBox\Desktop\OpenCPN-4.4.0\build\expat.lib(expat.dll)	1	
Error	LNK2005	_XML_SetDefaultHandler already defined in wxexpatd.lib(xmlparse.obj)	opencpn	C:\Users\virtualBox\Desktop\OpenCPN-4.4.0\build\expat.lib(expat.dll)	1	
Error	LNK2005	_XML_SetUnknownEncodingHandler already defined in wxexpatd.lib(xmlparse.obj)	opencpn	C:\Users\virtualBox\Desktop\OpenCPN-4.4.0\build\expat.lib(expat.dll)	1	
Error	LNK2005	_XML_SetUserData already defined in wxexpatd.lib(xmlparse.obj)	opencpn	C:\Users\virtualBox\Desktop\OpenCPN-4.4.0\build\expat.lib(expat.dll)	1	
Error	LNK2005	_XML_Parse already defined in wxexpatd.lib(xmlparse.obj)	opencpn	C:\Users\virtualBox\Desktop\OpenCPN-4.4.0\build\expat.lib(expat.dll)	1	
Error	LNK2005	_XML_GetErrorCode already defined in wxexpatd.lib(xmlparse.obj)	opencpn	C:\Users\virtualBox\Desktop\OpenCPN-4.4.0\build\expat.lib(expat.dll)	1	
Error	LNK2005	_XML_GetCurrentLineNumber already defined in wxexpatd.lib(xmlparse.obj)	opencpn	C:\Users\virtualBox\Desktop\OpenCPN-4.4.0\build\expat.lib(expat.dll)	1	
Error	LNK2005	_XML_ParserFree already defined in wxexpatd.lib(xmlparse.obj)	opencpn	C:\Users\virtualBox\Desktop\OpenCPN-4.4.0\build\expat.lib(expat.dll)	1	
Error	LNK2005	_XML_ErrorString already defined in wxexpatd.lib(xmlparse.obj)	opencpn	C:\Users\virtualBox\Desktop\OpenCPN-4.4.0\build\expat.lib(expat.dll)	1	
Error	LNK1169	one or more multiply defined symbols found	opencpn	C:\Users\virtualBox\Desktop\OpenCPN-4.4.0\build\Debug\opencpn.exe	1	
Error	LNK2001	unresolved external symbol "int const wxEVT_DOWNLOAD_EVENT" (?wxEVT_DOWNLOAD_EVENT@@3HB)	chartdldr_pi	C:\Users\virtualBox\Desktop\OpenCPN-4.4.0\build\plugins\chartdldr_pi\chartdldr_pi.obj	1	

```

CaptPat, 2017-09-11:
did you resolve this?  i think you have generated wxWidgets from source and they contain their own copy of expat - not sure what version, but that accounts for the multiple definitions.  The solution appears to be to subset wxWidgets and not use their wxexpatd.lib in the link step - in buildwin in the opencpn tree there is an expat-2.1.0 reference that should contain the expat.dll and other files.
Not sure about that last error - I am getting it with 3.0.3 wxWidgets also... don';t think i saw it with 3.1... but i was told to build with 3.0.2 or 3.0.3...  if you got past the last one, i'd be interested in how...! 
thanks!

pavelpasha, 2017-09-11:
I changed VS2015 to VS2013 version. I don`t know why, but it solved the problem.

nohal, 2017-12-13:
Closing as it is not reproducible with a correctly done build.

closed 2017-12-13 by nohal

Rasbats, 2018-08-10:
Tried to compile OpenCPN 4.8.6 using OpenCPN_buildwin-4.8.4 and VS2013. WxWidgets 3.0.2. Needed to make some additions to the linker files and remove the link to wxWidgets 'expat.lib'. The 'expat.lib' from buildwin is used.
The remaining problem is compiling chartdldr_pi.
Code:
7>chartdldr_pi.obj : error LNK2001: unresolved external symbol "int const wxEVT_DOWNLOAD_EVENT" (?wxEVT_DOWNLOAD_EVENT@@3HB)

Tried '#include "wx/dlimpexp.h"' in 'chartdldr_pi.h' but this did not work.

Hakansv, 2018-08-10:
For info are the same failures valid for present master build 4.99.0 using Win10, VS2017 and wx 3.1.1
Since I'm not clever enough to learn CMakeLists.txt to exclude wxexpat.lib and wxexpatd.lib I simply deleted these files in the wx lib-library. 

But the chartdldr error is still there. (I once tried the same as Rasbats :-))