OpenCPN/OpenCPN #779: Add a "description" field for routes

emilecantin, 2017-02-15:
I'm currently planning my summer navigation, and I've encountered a few times where I'd like to attach some form of notes to a route. (E.g "Leave 2h after high tide" or similar"). I know I could attach them to a waypoint, but sometimes it applies to the whole route, so it would make more sense there.

I see that tracks already have that field, and it's supported in the GPX standard, so no issues there. (http://www.topografix.com/GPX/1/1/#type_rteType)

emilecantin, 2017-02-15:
Apparently the Route model already has it: https://github.com/OpenCPN/OpenCPN/blob/18a5312bd0d604d202d363edb6758c2b8e9a17f8/include/Route.h#L113

nohal, 2018-02-04:
Too late for 4.8.2, I am closing this as #968 supersedes it

closed 2018-02-04 by nohal