OpenCPN/OpenCPN #820: OpenCPN 4.6.0 in Ubuntu - toolbar

linbrown, 2017-03-27:
I ran OpenCPN 4.6.0 in Ubuntu 14.04. For some reasons, I cannot see nor have access to the toolbar (a bunch of controls shown as icons). OpenCPN 4.4.0 runs fine with the toolbar. OpenCPN 4.6.0 is fine on Windows. Could there be a problem or how can I get access to the toolbar? Is there a setting in the configuration file somehow to make OpenCPN to show the toolbar by default?

Thanks!

kerpal-j, 2017-05-13:
i had the same problem with opencpn 4.6.1 on ubuntu 16.04.2. ppa install and compiling from src (gtk2) both led to no toolbar. trying to access menu items also failed with:

Gtk-CRITICAL **: IA__gtk_widget_set_size_request: assertion 'width >= -1' failed

i tried changing a few options in opencpn.conf but no dice. i'm back on 4.4.0 for the time being as well.

linbrown, 2017-05-15:
Try to use "ALT" key (https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:keyboard_shortcuts).

nohal, 2018-01-15:
Reproducible for anybody with the current versions?
FWIW it never was for me on what currently is something like Ubuntu 16.04 with Unity.

nohal, 2018-02-04:
The question stands - reproducible for anybody with some current, not completely minor, distribution?

linbrown, 2018-08-13:
The problem seems to still persist, even in OpenCPN 4.8.4 and 4.8.6 under Ubuntu (I did not try other Linux OS). Apparently sometimes works and sometimes not. It seems that ocpnToolbarDialog was created with very small dimension when not visible. 

**ocpnToolbarDialog characteristics when no show:**

xwininfo: Window id: 0x1000292 "ocpnToolbarDialog"

  Absolute upper-left X:  0
  Absolute upper-left Y:  0
  Relative upper-left X:  0
  Relative upper-left Y:  0
  Width: 2
  Height: 1
  Depth: 24
  Visual: 0x21
  Visual Class: TrueColor
  Border width: 0
  Class: InputOutput
  Colormap: 0x20 (installed)
  Bit Gravity State: NorthWestGravity
  Window Gravity State: NorthWestGravity
  Backing Store State: NotUseful
  Save Under State: no
  Map State: IsUnMapped
  Override Redirect State: no
  Corners:  +0+0  -1022+0  -1022-767  +0-767
  -geometry 2x1+0+0

**ocpnToolbarDialog characteristics when it does show in OpenCPN:**

xwininfo: Window id: 0x134330d (has no name)

  Absolute upper-left X:  289
  Absolute upper-left Y:  32
  Relative upper-left X:  289
  Relative upper-left Y:  32
  Width: 64
  Height: 23
  Depth: 24
  Visual: 0x21
  Visual Class: TrueColor
  Border width: 0
  Class: InputOutput
  Colormap: 0x20 (installed)
  Bit Gravity State: NorthWestGravity
  Window Gravity State: NorthWestGravity
  Backing Store State: NotUseful
  Save Under State: no
  Map State: IsUnMapped
  Override Redirect State: no
  Corners:  +289+32  -671+32  -671-713  +289-713
  -geometry 64x23+289+32

Thank you!

rgleason, 2018-08-13:
Put in FAQ for others with the problem
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:faq#toolbar_is_not_visible_in_opencpn_484_and_460_under_ubuntu

nohal, 2019-02-12:
@linbrown still relevant for current O5 beta codebase?

nohal, 2019-03-12:
If no new info appears and as nobody else was ever able to reproduce, will be closed on the next sweep.

kerpal-j, 2019-03-21:
i just compiled 4.99.1520 on debian 9 and the toolbar is visible and working well. thanks for the fix.

closed 2019-03-21 by nohal