OpenCPN/OpenCPN #826: Android version - segment violation

beaumontk, 2017-03-29:
In chart1.cpp at about line 2076
    gFrame->Show( TRUE );

seems to cause this crash (from logcat):
F/libc    ( 5832): Fatal signal 11 (SIGSEGV), code 2, fault addr 0xa27d2fe0 in tid 5859 (Thread-529)

I configured wxqt --with-opengl
I think that the fault is due to the qt version of toplevel.cpp being used in the build.
wxTopLevelWindowQt::Show doesn't exist! (Not very much seems to exist!)
But it exists in the gtk and gtk1 versions

So I commented out line 2076, but then got another similar crash.
Probably from another missing function.

What config should I be using? OpenCPN seems to need --with-opengl.

crazysong, 2018-03-16:
cd /home/janson/OpenCPN/wxqt/wxWidgets-3.1.0/build_android_gl
export PKG_CONFIG_PATH=/home/janson/Qt5.5.1/5.5/android_armv7/lib/pkgconfig
export CPPFLAGS=-D__ANDROID__
export PATH=/home/janson/Android/android-ndk-r10e/mytoolchain/bin:$PATH
export CC=arm-linux-androideabi-gcc
export CXX=arm-linux-androideabi-g++
../configure --with-qt --build=x86_64-unknown-linux-gnu --host=arm-linux-androideabi --enable-compat28 --disable-shared --disable-arttango --enable-image --disable-dragimage --with-libtiff=no --disable-baseevtloop --disable-xrc --disable-cmdline --disable-miniframe --disable-mdi --enable-debug --disable-stc --disable-ribbon --disable-propgrid --disable-datepick --enable-xlocale --enable-intl --with-opengl --enable-selectloop
make

cd /home/janson/OpenCPN/wxqt/OpenCPN-android/build_android
cmake -DUSE_GARMINHOST=OFF -D_wx_selected_config=androideabi-qt -DCMAKE_TOOLCHAIN_FILE=../buildandroid/cross_build_android.cmake ..
make

cd /home/janson/OpenCPN/wxqt/OpenCPN-android/build_android
export ANDROID_SDK_ROOT=/home/janson/Android/Sdk
export ANDROID_NDK_ROOT=/home/janson/Android/android-ndk-r10e
export PATH=/home/janson/Android/android-ndk-r10e/mytoolchain/bin:$PATH

export ANDROID_NDK_TOOLCHAIN_VERSION=4.9
export ANDROID_NDK_PLATFORM=android-19

/home/janson/Qt5.5.1/5.5/android_armv7/bin/qmake -makefile ../buildandroid/opencpn.pro -o Makefile.android -r -spec android-g++ CONFIG+=debug

make -f Makefile.android
make -f Makefile.android install INSTALL_ROOT=./apk_build
/home/janson/Qt5.5.1/5.5/android_armv7/bin/androiddeployqt --input ./android-libopencpn.so-deployment-settings.json  --output ./apk_build --android-platform android-19 --deployment bundled

this is mine, but i was failed at    -----   "make -f Makefile.android" ----

beaumontk, 2018-03-17:
It seems that android version went private! - not supported anymore without payment

nohal, 2018-03-17:
Not true, all the code is in https://github.com/OpenCPN/OpenCPN/tree/android

beaumontk, 2018-03-17:
Hi Nohal,
I started this thread a year ago! There had been no reply until now.
Has this problem been fixed?

nohal, 2018-03-17:
I have no idea. And every time I see an exclamation mark add another year to my personal plan to look into your issue.

beaumontk, 2018-03-17:
So it's not supported any more! The author went private!

nohal, 2018-03-17:
I probably don't understand what private means for you, but it really does not matter I think.

beaumontk, 2018-03-17:
By private I mean that the Android version is NOT open source anymore. If you pay, you get support. It's very clear to me. I have no objection to this, but it should be clear on the open source version.

bdbcat, 2018-03-17:
OpenCPN for Android is fully supported, and is still open-source. 

There has not been much interest in building OpenCPN for Android from source... The github repo is a little out-of-date, and there is work to be done on the README, etc. to make the image buildable from bare-metal.  This is on the list of things to do as resources allow.

beaumontk, 2018-03-17:
Hello bdbcat,
I guess that you are the author? The README was VERY out of date a year ago! I think that it should not take much work to correct it, if the app is truly buildable for Android. Have the problems I reported above (a year ago) been fixed now?

rgleason, 2018-03-17:
beaumontk, give Dave a break, he's a busy guy and try to be polite like you're talking to a friend, rather than demanding.., it'll take you much further.. !

beaumontk, 2018-03-18:
rgleason, I just would like to know the truth about this situation. If you read the whole of this issue, you might just understand my doubts.

transmitterdan, 2018-03-18:
Mr. Beaumont - you have been given the truth. The source code is open and available to anyone. Help with each unique build environment is provided by volunteers. So my advice to you is to ask for help nicely. And remember that this is a volunteer effort. So your help is appreciated. Yelling at people not so much.

beaumontk, 2018-03-18:
transmitterdan, your reply is bullshit. Firstly, I don't need help. Secondly, I started this issue more than a year ago. There had been NO reply until crazysong posted a couple of days ago. Thirdly, I lost interest in this app a year ago. Fourthly, and most importantly, there is NO OPEN-SOURCE version for Android so far as I know. AND I am  NOT yelling, just writing the facts as far as I know them.

rgleason, 2018-03-18:
"Firstly, I don't need help. "
Ok. What's the big deal then?  You don't need help, end of story. You are perfectly capable to figure out what is wrong with your configuration. Peace be with you.

beaumontk, 2018-03-19:
See my comment after crazysong. Isn't it clear that I was advising him/her that the Android version is not supported. The big deal is for crazysong, NOT me!

crazysong, 2018-03-22:
i have built an apk package，this version is ok，https://github.com/OpenCPN/OpenCPN/tree/android

but program crashed immediately when it start up.

i dont know what happened, maybe version of wxwidgets was wrong.

crazysong, 2018-03-22:
1.安装SDK（r24.4.1）
	r25以后不支持android命令。安装API level 21平台。
2.安装NDK（r10e）
3.安装qt5.5.1

=============================================================

4.编译WXQT
	a.下载源代码
	3.1.0
	b.新建编译路径
	cd /home/janson/OpenCPN/wxqt/wxWidgets-3.1.0
	mkdir build_android_gl
	c.修改configure.in
	found_gl=1
        LDFLAGS_GL="-L/usr/lib/x86_64-linux-gnu"
        OPENGL_LIBS="-lGLESv2"
[image]
[image]
	cd /home/janson/OpenCPN/wxqt/wxWidgets-3.1.0
	aclocal -I build/aclocal/
	autoconf
	d.修改代码
	qt/glcanvas.cpp、common/glcmn.cpp
[image]
[image]
[image]
[image]
**在/home/janson/OpenCPN/wxqt/wxWidgets-3.1.0/include/wx/qt/private/目录加入自有头文件**
setup.h.in 文件
设置wxUSE_GRAPHICS_CONTEXT为1
设置wxUSE_GLCANVAS为1
glcmn.cpp还是把bool wxGLCanvasBase::ParseAttribList方法给注释掉了，因为wx的qt实现分支确实没有实现glcanvas.h里面定义的一些方法。
	e.编译
cd /home/janson/OpenCPN/wxqt/wxWidgets-3.1.0/build_android_gl

export PKG_CONFIG_PATH=/home/janson/Qt5.5.1/5.5/android_armv7/lib/pkgconfig
export CPPFLAGS=-D__ANDROID__
export PATH=/home/janson/Android/android-ndk-r10e/mytoolchain/bin:$PATH
export CC=arm-linux-androideabi-gcc
export CXX=arm-linux-androideabi-g++
../configure --with-qt --build=x86_64-unknown-linux-gnu --host=arm-linux-androideabi --enable-compat28 --disable-shared --disable-arttango --enable-image --disable-dragimage --with-libtiff=no --disable-baseevtloop --disable-xrc --disable-cmdline --disable-miniframe --disable-mdi --enable-debug --disable-stc --disable-ribbon --disable-propgrid --disable-timepick --disable-datepick --enable-xlocale --enable-intl --with-opengl --enable-selectloop
make

==================================================================

5.编译OpenCPN-Android
	a.下载源代码
	android分支
	b.新建编译目录
	cd /home/janson/OpenCPN/wxqt/OpenCPN-android
	mkdir build_android
	c.修改{opencpn_root}/buildandroid/ cross_build_android.cmake
	指定wxQt_Base、wxQt_Build、Qt_Base等
	d.使用cmake配置项目编译
	cmake -DUSE_GARMINHOST=OFF -D_wx_selected_config=androideabi-qt -DCMAKE_TOOLCHAIN_FILE =../buildandroid/cross_build_android.cmake ..
	e.编译
	make

6.编译依赖库
cd /home/janson/OpenCPN/wxqt/OpenCPN-android/buildandroid/assetbridge
export PATH=/home/janson/Android/android-ndk-r10e/mytoolchain/bin:$PATH
export CC=arm-linux-androideabi-gcc
export CXX=arm-linux-androideabi-g++
export NDK_PROJECT_PATH=/home/janson/OpenCPN/wxqt/OpenCPN-android/buildandroid/assetbridge
/home/janson/Android/android-ndk-r10e/ndk-build
7.打包
cd /home/janson/OpenCPN/wxqt/OpenCPN-android/build_android
export ANDROID_SDK_ROOT=/home/janson/Android/android-sdk-linux
export ANDROID_NDK_ROOT=/home/janson/Android/android-ndk-r10e
export PATH=/home/janson/Android/android-ndk-r10e/mytoolchain/bin:$PATH
export ANDROID_NDK_PLATFORM=android-21

/home/janson/Qt5.5.1/5.5/android_armv7/bin/qmake -makefile ../buildandroid/opencpn.pro -o Makefile.android -r -spec android-g++ CONFIG+=debug

make -f Makefile.android
make -f Makefile.android install INSTALL_ROOT=./apk_build
/home/janson/Qt5.5.1/5.5/android_armv7/bin/androiddeployqt --input ./android-libopencpn.so-deployment-settings.json  --output ./apk_build --android-platform android-21 --deployment bundled

rgleason, 2018-03-22:
If this is the most current version, I believe it should be wxWidgets 3.1.1 which is what the OS linux, windows and macos versrion 4.99 uses.

crazysong, 2018-03-22:
@rgleason you mean i can compile android version using "code of version 4.8.2" ?
[image]

rgleason, 2018-03-22:
Have you been compiling the head?  That is probably more risky, particularly for Android.
(Of course, I know very little about the process. I cannot fathom why someone would want to go through this trouble when for a very small sum you get lifetime updates and maintenance!  What a deal.)

Anyway I think you can try compiling v4.8.2 as that version is what is on my official OpenCPN app  android phone,  Good luck, this is not easy, and I know that even the developers have challenges compiling for Android and there are details and skills involved that are developed from years of work and knowledge, so if you succeed, congratulations.

crazysong, 2018-03-22:
@rgleason hahahahaha, I'll try , thank you for your help , and maybe i need more help next.

rgleason, 2018-03-22:
Maybe you can help us all with programming and improving this program too? It would certainly be appropriate and fitting to get involved with the "creation" end of the process and lending a "strong shoulder to the wheel"!

crazysong, 2018-03-23:
@rgleason  there are so many bugs when I compile android app with v4.8.2，it make me confused --!

crazysong, 2018-03-23:
@rgleason Finally I have compiled and run the app using OpenCPN-android & wxwidgets v 3.1.1, thank you for your advise !

[image]
not easy !  thank God! thank God! thank God!

beaumontk, 2018-03-23:
@crazysong Can you post a list of your steps to compile? Maybe edit the README and post here.

rgleason, 2018-03-23:
Congratulations.  Bdbcat is probably too busy programming, or taking a well deserved rest.

crazysong, 2018-03-23:
@beaumontk ok, I will spend some time on it.

beaumontk, 2018-03-24:
Great. I'm doing other things now, but it will be useful for anybody else who likes the idea. My skills were not sufficient to solve the problems. Did you find this? "wxTopLevelWindowQt::Show doesn't exist!"

nohal, 2018-03-24:
Isn't it inherited from `wxWindow::Show()`? How exactly they remove it for Qt? I really don't see anything like that in the code.

crazysong, 2018-03-25:
@nohal hi sir , is my wxQtGesture.h out of date?
[image]

crazysong, 2018-03-25:
@beaumontk I can't find "wxTopLevelWindowQt::Show" anywhere too.

beaumontk, 2018-03-25:
Sorry, this is going to be long winded.
I raised this issue a year ago, so it's very difficult to remember, but my thinking was as follows:

chart1.cpp contains:
'''
MyFrame                   *gFrame;
gFrame = new MyFrame( NULL, myframe_window_title, position, new_frame_size, app_style ); //Gunther
// My frame constructor
MyFrame::MyFrame( wxFrame *frame, const wxString& title, const wxPoint& pos, const wxSize& size,
        long style ) :
'''

beaumontk, 2018-03-25:
Sorry, this is going to be long winded and I don't know how to make it look neat.
I raised this issue a year ago, so it's very difficult to remember, but my thinking was as follows:

chart1.cpp contains:
MyFrame                   *gFrame;
gFrame = new MyFrame( NULL, myframe_window_title, position, new_frame_size, app_style ); //Gunther
// My frame constructor
MyFrame::MyFrame( wxFrame *frame, const wxString& title, const wxPoint& pos, const wxSize& size,
        long style ) :

So gFrame is an instance of wxFrame.

My build was congigured with qt

In wxqt/wxWidgets/src/qt/frame.cpp

wxFrame::wxFrame( wxWindow *parent, wxWindowID id, const wxString& title,
        const wxPoint& pos, const wxSize& size, long style, const wxString& name )
{
    Create( parent, id, title, pos, size, style, name );
}

wxFrame is defined in /home/kb/wxqt/wxWidgets/include/wx/qt/frame.h with wxWindow as parent
qt/frame.h includes /home/kb/wxqt/wxWidgets/include/wx/frame.h
which includes wx/toplevel.h
which declares wxTopLevelWindow class
wxTopLevelWindow is found in wxqt/wxWidgets/src/qt/toplevel.cpp

chart1.cpp calls gFrame->Show
BUT toplevel.cpp DOES NOT contain a show method, hence the crash with a segment violation

This was my trace:
I/app_name(11004): ----------------------------------------
I/app_name(11004):   # 0: 0xa32f714c  _Z16captureBacktracePPvj
I/app_name(11004):   # 1: 0xa32f7328  _Z17backtraceToLogcatv
I/app_name(11004):   # 2: 0xa3115cd8  _ZN7wxFrame6CreateEP8wxWindowiRK8wxStringRK7wxPointRK6wxSizelS4_
I/app_name(11004):   # 3: 0xa3115b48  _ZN7wxFrameC2EP8wxWindowiRK8wxStringRK7wxPointRK6wxSizelS4_
I/app_name(11004):   # 4: 0xa2c8fdac  _ZN7MyFrameC2EP7wxFrameRK8wxStringRK7wxPointRK6wxSizel
I/app_name(11004):   # 5: 0xa2c932ec  _ZN5MyApp6OnInitEv
I/app_name(11004):   # 6: 0xa2ca3004  _ZN16wxAppConsoleBase10CallOnInitEv
I/app_name(11004):   # 7: 0xa2c68550  main
I/app_name(11004):   # 8: 0xa3777602  
I/app_name(11004):   # 9: 0xb6dff038  
I/app_name(11004):   #10: 0xb6dfd008  
I/app_name(11004): ----------------------------------------

beaumontk, 2018-03-25:
Sorry, this is going to be long winded and I don't know how to make it look neat.
I raised this issue a year ago, so it's very difficult to remember, but my thinking was as follows:

chart1.cpp contains:
MyFrame                   *gFrame;
gFrame = new MyFrame( NULL, myframe_window_title, position, new_frame_size, app_style ); //Gunther
// My frame constructor
MyFrame::MyFrame( wxFrame *frame, const wxString& title, const wxPoint& pos, const wxSize& size,
        long style ) :

So gFrame is an instance of wxFrame.

My build was configured with qt

In wxqt/wxWidgets/src/qt/frame.cpp

wxFrame::wxFrame( wxWindow *parent, wxWindowID id, const wxString& title,
        const wxPoint& pos, const wxSize& size, long style, const wxString& name )
{
    Create( parent, id, title, pos, size, style, name );
}

wxFrame is defined in /home/kb/wxqt/wxWidgets/include/wx/qt/frame.h with wxWindow as parent
qt/frame.h includes /home/kb/wxqt/wxWidgets/include/wx/frame.h
which includes wx/toplevel.h
which declares wxTopLevelWindow class
wxTopLevelWindow is found in wxqt/wxWidgets/src/qt/toplevel.cpp

chart1.cpp calls gFrame->Show
BUT toplevel.cpp DOES NOT contain a show method, hence the crash with a segment violation

This was my trace:
I/app_name(11004): ----------------------------------------
I/app_name(11004):   # 0: 0xa32f714c  _Z16captureBacktracePPvj
I/app_name(11004):   # 1: 0xa32f7328  _Z17backtraceToLogcatv
I/app_name(11004):   # 2: 0xa3115cd8  _ZN7wxFrame6CreateEP8wxWindowiRK8wxStringRK7wxPointRK6wxSizelS4_
I/app_name(11004):   # 3: 0xa3115b48  _ZN7wxFrameC2EP8wxWindowiRK8wxStringRK7wxPointRK6wxSizelS4_
I/app_name(11004):   # 4: 0xa2c8fdac  _ZN7MyFrameC2EP7wxFrameRK8wxStringRK7wxPointRK6wxSizel
I/app_name(11004):   # 5: 0xa2c932ec  _ZN5MyApp6OnInitEv
I/app_name(11004):   # 6: 0xa2ca3004  _ZN16wxAppConsoleBase10CallOnInitEv
I/app_name(11004):   # 7: 0xa2c68550  main
I/app_name(11004):   # 8: 0xa3777602  
I/app_name(11004):   # 9: 0xb6dff038  
I/app_name(11004):   #10: 0xb6dfd008  
I/app_name(11004): ----------------------------------------

beaumontk, 2018-03-25:
Correction
This was my trace:
I/app_name(11004): ----------------------------------------
I/app_name(11004):   # 0: 0xa32f714c  _Z16captureBacktracePPvj
I/app_name(11004):   # 1: 0xa32f7328  _Z17backtraceToLogcatv
I/app_name(11004):   # 2: 0xa317f954  _ZN18wxTopLevelWindowQt6CreateEP8wxWindowiRK8wxStringRK7wxPointRK6wxSizelS4_
I/app_name(11004):   # 3: 0xa3115d6c  _ZN7wxFrame6CreateEP8wxWindowiRK8wxStringRK7wxPointRK6wxSizelS4_
I/app_name(11004):   # 4: 0xa3115b48  _ZN7wxFrameC2EP8wxWindowiRK8wxStringRK7wxPointRK6wxSizelS4_
I/app_name(11004):   # 5: 0xa2c8fdac  _ZN7MyFrameC2EP7wxFrameRK8wxStringRK7wxPointRK6wxSizel
I/app_name(11004):   # 6: 0xa2c932ec  _ZN5MyApp6OnInitEv
I/app_name(11004):   # 7: 0xa2ca3004  _ZN16wxAppConsoleBase10CallOnInitEv
I/app_name(11004):   # 8: 0xa2c68550  main
I/app_name(11004): ----------------------------------------

nohal, 2018-03-25:
@beaumontk  Why should `toplevel.cpp` contain any **code** for the Show method at all? That method is inherited from the `wxWindow` class. If it was missing, you would not be able to link the binary and would never get to running it and a crash of any kind as you would simply have no binary to start. I'm afraid you are looking in a completely wrong direction. But hard to be sure as we do not know what exactly you were doing before you got completely uninterested in OpenCPN.

@crazysong I really don't know if your `wxQtGesture.h` is out of date as I don't see any code of yours anywhere. We also do not know what you are doing exactly.

crazysong, 2018-03-26:
@nohal I am compiling app with the branch 'android' of OpenCPN, & Qt5.5.1 & wxWidgets '3.1.1'.
And  in glchartcanvas.cpp , there are some gesture event response method,  we used private header 'wxQtGesture.h', so I download these headers from https://github.com/bdbcat/wxWidgets.git.

such as this method below,
glChartCanvas::OnEvtPinchGesture(wxQT_PinchGestureEvent &event),  
m_pinchStart = **event.GetCenterPoint();**
it used **wxQT_PinchGestureEvent.GetCenterPoint()**
but our wxQtGesture.h hasn't declare GetCenterPoint(), are these private headers out of date ??

thanks!

nohal, 2018-03-26:
It certainly worked fine with the rest of the code from https://github.com/bdbcat/wxWidgets.git when I tried. I have of course no idea if it works without any changes if you mix this header with a completely different version of wxWidgets. As there were quite a few changes related to the gestures up to wx3.1.1, I would presume not. Which seems to be in line with what your picture above shows.

bdbcat, 2018-03-26:
1.  The source on github for OpenCPN Android is known to be out of date.

2.  The behaviour and logic of wxqt on wx3.1.1 is untested with OpenCPN for Android.  I'm sure that many of the recent local/private changes for OCPN did not make it upstream into 3.1.1.  I'm actually quite surprised that it links and starts at all with vanilla wxqt 3.1.1.  But crazysong showed that it does, to some point, so this should make life easier going forward....

3.  These build problems are not something that I will be able to resolve in the short term.  Other critical issues are pre-emptive of my time.  But I will get back to this soon.  Please be patient, and feel free to experiment.

Dave

crazysong, 2018-03-26:
@bdbcat oh, I am sorry to hear about that.
1. you mean that the OpenCPN-android code I used was out of date? Can you tell me how to get the right code, the branch 4.8.2?
2. Now, I have compile the android app and run it in android 5, there are some problem:
all gestures are disabled and crashed frequently.

bdbcat, 2018-03-26:
" Can you tell me how to get the right code?"
No, not right now.  Sorry.

crazysong, 2018-03-26:
[image]
So why it would be out of date, it seems to be constantly supported ? isn't it ?

crazysong, 2018-03-26:
@bdbcat oh, haha , so I must give up to compile ocpn by myself, right?

nohal, 2018-03-26:
As you can see from your picture, it is currently missing code created in the last 3 months. Nobody told you you must give up, and @bdbcat clearly stated he will as usually update the android branch as time permits.

crazysong, 2018-03-26:
but the code is unable  to be compile nicely, if I want to add something for myself used, I have no idea what I can do.

nohal, 2018-03-26:
Then help to make it "compile nicely" by actually contributing instead of complaining. Or stop complaining and wait until others have time and motivation to do it. Or switch to a platform with broader developer base and better support, like Linux, macOS or Windows.

crazysong, 2018-03-26:
Yes,you are right,but you know I need a correct base line

发自我的 iPhone

nohal, 2018-03-26:
You have a readily available "correct baseline" in the master branch, where I can personally help you with any build related problems you may encounter on any of the supported desktop platforms. It will sure help you to familiarize yourself with the codebase while you wait for the android branch to be updated and provide more precise build instructions. Looking forward for your pull requests.

crazysong, 2018-03-26:
@nohal thank you very much!  I want to do something base on it and I hope I can share them to everybody. Maybe My skills were not sufficient to do this, I'll try, haha.

crazysong, 2018-04-08:
@beaumontk hi~ guy, the building process take me some time, it's no easy, but everything is ok now.
you can build OCPN according to this DOC.

Thank to everyone for your helps
[README.docx](https://github.com/OpenCPN/OpenCPN/files/1887469/README.docx)

crazysong, 2018-04-11:
[image]
[image]

crazysong, 2018-04-14:
maybe someone will need them
@nohal @bdbcat 

draw the real size of ais target instead of the triangle symbol if it can be show as real size
[image]

select ais target logic changed, select the target who's closest to touch point  or the touch point is in the  real scope of target
[image]

ownship style
[image]

ais target list, add filter 
[image]

and I am  a fresh hand of github , how to make these source opened

crazysong, 2018-04-14:
[image]

nohal, 2018-04-14:
https://help.github.com/articles/fork-a-repo/
https://help.github.com/articles/creating-a-pull-request/

nohal, 2018-04-14:
FWIW, OpenCPN can draw real size of AIS targets for years, but I don't use the Android version.

crazysong, 2018-04-14:
yes, android version draw the real size at left top of screen
it doesn't tranlate to target point

nohal, 2018-04-14:
Don't understand. I sure will looking at the pull request though.

rgleason, 2018-04-16:
Note that rooiedirk created an AIS setting for "Attenuate less important targets" which when used will change the scale of AIS targets. AIS is used at various scales and conditions: Harbor, Coastal, Ocean, etc.
These modifications may be best for harbor. The attenuate setting may be best for coastal scale. These two settings may be exclusive however.

closed 2021-04-12 by bdbcat