OpenCPN/OpenCPN #841: Grib_pi

rgleason, 2017-05-26:
I am not sure if this is where this should be posted, but the plugin is included with the program (I will do Tracker too)
1. Opencpn Grib_pi Downloads using zygrib username and password is no longer available, so this download feature appears to be broken and should be fixed.
2. Opencpn Grib_pi Downloads using Saildocs with Rtofs request results in 24 hr gribs even when 12 hr is requested. Additionally 3hr Rrtof current gribs are available so that should work too, but Saildocs grib server has been affected by the changes at NOAA to using the RTOFS Global Model, thus breaking the Saildocs Rtofs download.
3. Since Grib_pi has been updated to grib2 and since we can load two grib files now, it would be truely wonderful to update this handy and essential plugin for improved downloading.
4. I have found the "walking or moving grib" feature invaluable for reducing the size data downloads and hope that at some point download of Rtofs currents would work in the same way with 3hr resolution.

rgleason, 2017-12-18:
1. I found a resolution to this item. The zygrib website changed the login location to get the special code. I've tested the process and it does work fine, once you know the login location.  It is no longer the first page of the website.  URL http://www.zygrib.org/index.php?page=gribauto

nohal, 2018-01-15:
Rick...
Please, one issue per problem/enhancement request...
So to recapitulate:
1) Does not exist? Or is it some real issue?
2) Is it our issue? Or Saildocs'? Or not an issue?
3) Is covered by https://github.com/OpenCPN/OpenCPN/issues/967 now
4) Do such data exist? If so, where and how to obtain? OR is it "Oh, it would be nice if that thing existed?"
Sorry for not doing investigation through the link, but this thing should save us time, not generate more work.

rgleason, 2018-01-15:
1. Login has moved to a back page, and exists.  I put a note in the manual.
2. Saildocs issue I guess.  24 hr gribs even when 12 hr is
3. Improved downloading. 967  - thank you.
4. Rtofs - is getting better, still does not include tidal currents and it should. Interval is down to 3hrs. which is almost good enough, but for tidal coastline it needs to be 1 hr.

Thanks, I am closing this.

closed 2018-01-15 by rgleason