OpenCPN/OpenCPN #851: Please add a Debian/REQUIREMENTS

pcav, 2017-06-20:
I currently cannot compile because I do not find the appropriate wx package:

```
CMake Error at /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
```

This will make it easier to self compile the package.
Thanks.

pcav, 2017-06-24:
In fact, detailed instructions are here:
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_linux
Just Another Effect Of Duplicating Information ;)

nohal, 2018-01-15:
The launchpad publishing stuff we use to create the DEBs lives in https://github.com/nohal/launchpad and there is currently no plan to merge it to the base tree.

For the adventurous, there always is https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=538067

And you have already found the documentation...

pcav, 2018-01-16:
Pity, a proper package would increase the Debian user base enormously.

nohal, 2018-01-16:
Well, jump on the train and address some of the complaints the Debian mentors have...

nohal, 2018-01-16:
You sure don't have to be a programmer to for example clean up the issues they have with missing or outdated license headers in the various files.

pcav, 2018-01-16:
If I remember correctly, there are also issues with patched/embedded libraries, more tricky to solve.
Unfortunately the list of children to take care of is long, and time is scarce. I'll see what I can do, but be sure DebianGIS is an useful resource in case someone wants to do something.

nohal, 2018-01-16:
Nobody asks or expects you to fix everything. Same as nobody can ask or expect me to do it. Part of the complaints in the debian bug are also plain bogus where the mentors didn't bother to even look at what they talk about. All of that has to be handled though, somehow, by someone.

closed 2019-03-13 by bdbcat