OpenCPN/OpenCPN #853: Ubuntu 16.04 installing OpenCPN package

Rasbats, 2017-07-12:
Compiled a local version of OpenCPN on newly upgraded 14.04 to 16.04 machine (386). The package opens in the Software Centre but the following message appears:

Dependency is not satisfiable: libwxgtk3.0-0

Iibwxgtk3.0-0 has been removed from U 16.04. Not able to install it. Is there a workaround?

Mike

rgleason, 2017-07-18:
I believe I saw notes about this in the forum, Mike.
It explained how to add the lib. Perhaps you can search? I am on cell.. Should this info be added to the manual install instructions?

Rasbats, 2017-07-18:
Success: Note has been added to the Developer Manual.

[https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:compiling_linux#install_it_or_create_a_package](url)

Rasbats, 2017-07-18:
Editing CMakeLists.txt solved the problem.

closed 2017-07-18 by Rasbats