OpenCPN/OpenCPN #854: Windows 10 Creators update prevents shutting down OCPN with OpenGL

douwefokkema, 2017-07-16:
After the Microsoft update to Windows 10 version 1703 OpenCPN 4.6.1 (and 4.4) with OpenGL enabled will hang on closure. It can still be closed by minimizing O, but then the last position is not always saved (sometimes it is). All plugins are disabled. No problem when OpenGL is disabled before closing O.

transmitterdan, 2017-07-16:
I guess this is something peculiar to your Windows 10 machine. I have several Windows 10 machines with Creators update. No such issue has been seen. About all I can suggest is to look at the graphic drivers and see if new Windows 10 updates are available.

douwefokkema, 2017-07-17:
System has Intel HD Graphics 4000 with recent but not latest driver (can't install latest now, too risky on main boat system far from home). But it seems that the "exit click" is not past to O. On clicking the window exit x, screen gets dimmed but "OnExit()" is not called. O keeps running, I can still disable OpenGL, then normal exit follows. Issue clearly is outside O. Issue could indeed be the graphics driver or wxWidgets.

douwefokkema, 2017-07-24:
I did install the latest Intel display driver now. No difference. This problem is very irritating. 
So I looked into the code and found that O hangs on the wxYield(); in "void MyFrame::OnCloseWindow( wxCloseEvent& event )". I removed the wxYield(), close down is fine now. But I can't really understand what the wxYield() does, no proper internet now to sort this out.

transmitterdan, 2017-07-24:
wxYield() just lets the main event loop continue to process events. There are 2 calls to wxYield in OnCloseWindow(). Which one did you delete?

douwefokkema, 2017-07-25:
It was the first one that was the problem and that I deleted. Line 3442 of chart1.cpp.

transmitterdan, 2017-07-28:
I think I have found a solution to this. Have submitted a PR with what I think will work.

douwefokkema, 2017-08-03:
Solved in 4.6.22. Indeed removal of the wxYield() solved the issue.

closed 2017-08-03 by douwefokkema