OpenCPN/OpenCPN #857: Asserts in wxWidgets libraries

transmitterdan, 2017-07-24:
There are lots of lines of code in OpenCPN that cause wxWidgets to give a wxAssert in debug mode. This will get worse as we are eventually forced to use the latest version of wxWidgets which is much more pedantic than previous version.  I have started a branch in my fork to collect all of these "fixes" here:

[https://github.com/transmitterdan/OpenCPN/tree/wxFixes](url)

Would it be a good idea to have a tracking branch within the official repository to capture all of these "fixes"?  Testers could merge the master branch with this branch for testing. I have tried to test all the corrections I could find and they don't seem to cause any undesired side effects. I have also tested OpenCPN built with VS 2017 and the latest master branch of wxWidgets 3.1.1 with no wxAsserts or other ill effects noted. There is one annoying thing in wxWidgets that is easy to work around. See my fork of wxWidgets if interested here:

https://github.com/transmitterdan/wxWidgets/tree/OpenCPN

There are only 3 lines of code involving caching the output of _tzset() in time.cpp to avoid a bazillion calls which is slow under Windows. 

I think it would be good to think about the migration to latest wxWidgets and VS tools as there are several bugs in older wx releases that are fixed in the latest build of wxWidgets. And the new VS tools are vastly better than VS2010 or VS2013.

Dan

transmitterdan, 2018-02-07:
Pavel,

I have most of these fixed over in my fork. Please compare my wxFixes branch to OpenCPN master branch. If you agree these changes I can create a PR.

https://github.com/OpenCPN/OpenCPN/compare/master...transmitterdan:wxFixes?expand=1

A brief list of issues I have found/fixed:

1) WX3.1 will not expose the zlib interface going forward. Solution, include zlib sources for windows builds.
2) Sometimes the drag bar is not shown in dashboard so it is impossible to drag it out of a docked state. So add context menu item to undock.
3) Lots of runtime wxAsserts and compile time warnings.

Something I have found but not fixed:

Status bar and main chart window slightly overlap when using WX3.1. This gets fixed if I open the toolbox and close it via OK button. I think this is a bug in O not WX. No research done.

nohal, 2018-02-07:
Well, I would much like to see this broken into smaller, more manageable PRs - For example I am not at all sure that we want to include zlib source instead of using the zlib library we already do have in the dependency bundle. And it does not mean I say we don't, just that it will likely have to be evaluated, while the asserts are simply a direct merge...

nohal, 2018-02-07:
And yes, I would like to switch to C++11 and on Windows to VS 2017 as soon as possible. If some blocker won't appear...

transmitterdan, 2018-02-08:
I agree we can use zlib from dependency. I am unsure how to reconfigure the cmake files for grib_pi to use zlib1.dll instead of the zlib code found in wx. But it has to be done first else linking to latest master branch of wx3.1 fails.

I also agree to make smaller PR. If you can get grib_pi to build without linking to wx zlib code I can get the other edits for wx compile/run time issues into a much simpler PR.

transmitterdan, 2018-02-08:
Are we going to support XP with 5.0?

I already build and run with VS2017. I believe I am building an installation that will deploy and run on XP but I have no way to test whether it will really work as I no longer have an XP machine (really unsafe to use XP these days). I can describe my build process including how I build wx3.1 that should work under XP (vc141_xp toolset).

bdbcat, 2018-02-08:
tDan...
I can build and test on XP.

Dave

bdbcat, 2018-02-08:
Dan...
Of course there is no reason to actually build on XP, but I keep it around for archival posterity.
Dave

transmitterdan, 2018-02-08:
I am unsure if VS2017 can create a useable executable that will actually run on XP. If we switch to VS2017 and take advantage of C++11 will we be giving up XP support? I imagine some older users still use XP on board and will complain if support is dropped.

bdbcat, 2018-02-08:
Let us try and see what happens on the base.
Are there any changes required to build now on VS2017 and wx3.1 (skipping plugins due to zlib)?
Lets start there.

transmitterdan, 2018-02-08:
No, there are no changes needed to build OpenCPN/master to build on VS2017 using wx3.0.

The changes we are talking about here just remove compile time and runtime warnings. But the master branch will build and run when compiled with VS2017.

bdbcat, 2018-02-08:
OK, I'll try VS2017 tomorrow.

transmitterdan, 2018-02-08:
My cmake command line when using VS2017 (all on one line):

`cmake -Wno-dev -G "Visual Studio 15 2017" -T "v141_xp" -D CMAKE_CXX_FLAGS="/D_USING_V110_SDK71_ /MP /EHsc" -D CMAKE_C_FLAGS="/MP /D_USING_V110_SDK71_" CMAKE_EXE_LINKER_FLAGS=/SUBSYSTEM:WINDOWS",5.01" CMAKE_MODULE_LINKER_FLAGS=/SUBSYSTEM:WINDOWS",5.01" CMAKE_SHARED_MODULE_LINKER_FLAGS=/SUBSYSTEM:WINDOWS",5.01" ..
`

This is a batch file I use to build wxWidgets with VS2017

`ECHO ON

set compvers=vc141_xp
set comp=141_xp

rmdir %compvers%_mswuddll /s /q
rmdir %compvers%_mswudll /s /q

@echo deleting folder "..\..\lib\%compvers%_dll"
rmdir ..\..\lib\%compvers%_dll /s /q

@echo Delete the build output files from the last run.

del %compvers%x86_Debug.txt
del %compvers%x86_Release.txt

nmake /f makefile.vc BUILD=release SHARED=1 COMPILER_VERSION=%comp% CXXFLAGS=/D_USING_V110_SDK71_ CFLAGS=/D_USING_V110_SDK71_ LDFLAGS=/SUBSYSTEM:WINDOWS",5.01" >> %compvers%x86_Release.txt
nmake /f makefile.vc BUILD=debug SHARED=1 COMPILER_VERSION=%comp% CXXFLAGS=/D_USING_V110_SDK71_ CFLAGS=/D_USING_V110_SDK71_ LDFLAGS=/SUBSYSTEM:WINDOWS",5.01" >> %compvers%x86_Debug.txt
`

Then copy the dlls from wxWidgets\lib\vc141_xp_dll to buildwin\wxWidgets.

Hakansv, 2018-02-08:
I've a couple of XPs as well if needed for tests

canboat, 2018-02-08:
I can confirm that using the vc141_xp target to CMake will build executables that are compatible with XP, I use this elsewhere.

— 
Kees

nohal, 2018-02-08:
OK, I will move on with my favorite hobby of installing a new Windows builder and preparing the buildwin package with VS2017. Given there won't be any surprises it should not take more than a couple of days.

My notes to the subject collected so far are at https://github.com/nohal/OpenCPN/wiki/WX3.1-and-C----11-migration

transmitterdan, 2018-02-08:
I did not rebuild any of the buildwin tools/libs. I am using the same buildwin files as everyone else. But as I said, I have not tested my package on XP. If anyone would be willing to test I will create a W32 package built with VS2107 and latest WX master branch.

transmitterdan, 2018-02-08:
Here is a link to a 4.8.2 package build from lastest WX master and O master with my wx fixes added. If anyone can test on XP would be nice.

https://1drv.ms/u/s!Ahh4-68oWChagaNNUYb3MLIZSje1SA

Hakansv, 2018-02-08:
@transmitterdan 
I think you've to include msvcp140.dll in the installation package. The exe file can not run without it. 
And probably the msvcr140.dll as well. That may be next task if I solve the msvcd140 myself. 
I'll leave my XP system untouched for next test?
Håkan

nohal, 2018-02-08:
@Hakansv yes, please do not help it, we need a testing environment like a normal user does and we have to be installable and runnable on it without any manual intervention of course.

Hakansv, 2018-02-08:
@nohal 
Yes - clear.
And - please advice when your hobby project is ready for a novice test. I'll be happy to test a wxWidgets 3.1 and VS2017 switch procedure.

transmitterdan, 2018-02-08:
Ok, good information. I will add those to the package. That is why we are doing this.

transmitterdan, 2018-02-09:
Ok, I have added that msvcp140.dll to the package. Please check if this runs on XP or not.

https://1drv.ms/u/s!Ahh4-68oWChagaNNUYb3MLIZSje1SA

bdbcat, 2018-02-09:
Tdan...
Sorry, no start.
[image]

nohal, 2018-02-09:
The good old http://www.dependencywalker.com should show all the libs needed here. I have my VS2017-only builder ready and am something like 30% through the dependencies update/rebuild. Can go through the redistributables  when done as they need to be included in the bundle anyway...

transmitterdan, 2018-02-09:
I seem to recall that dll is one Microsoft says not to distribute. They want to send it out with Windows Update process. That way they can patch it for security holes without searching the entire hard drive for every copy. I'll try to find where I read about at some many months ago.

transmitterdan, 2018-02-09:
I misremembered. That one comes from vc_redistx86.exe. Lastest version is here:

https://www.visualstudio.com/downloads/ at the very bottom of the page.

nohal, 2018-02-11:
@bdbcat do we want to take our `ocpn_gltest1.exe` friend with us or will we leave him to die here? The `wxGLCanvas` API is significantly different in wx3.1 so he will need some bigger surgery to be even buildable...
Otherwise I think I'm "almost there" - everything except chart downloader already seems to work on real XP rebuilt with v141_xp toolchain. Often it is necessary to help the libraries to really be XP compatible manually, nobody seems to be testing these builds to really work anymore even in cases where it looks supported...

bdbcat, 2018-02-11:
@nohal 
ocpn_gltest1.exe has not been actually run since wx2.8, so many years ago.  I think we can safely retire him.

nohal, 2018-02-11:
I'm done. I think. I hope...
Testing installer with OpenCPN built with VS2017 on W10 with all the dependencies usable on XP with SP3 and higher: https://spideroak.com/storage/NZXWQYLM/shared/1092636-2-1026/opencpn_4.99.0_setup.exe?f6c22367f32f32389d9c49de00c50c41 (The link will autodestruct in 72 hours)

Hakansv, 2018-02-11:
O 4.99.0 tested so far on XP only. 
Seems to work fine. All inbound plugins are by a brief test working fine. No other plugins are are, as probably expected, not working. I've tested all my charts, KAP, ENC and CM93. oeSENC charts are consequently not working. 
OpenGl usage is fine and log file report as normal.
Zooming, moving around and chart quilt and type switch as normal.
More to test???
Håkan

nohal, 2018-02-11:
Chart downloader. I linked it with the system WinSSL in this build and suspect it to be too old on XP to work with modern TLS used by the servers today. (And we will need OpenSSL for `libwebsockets` anyway so I will probably switch `cURL` to use it as well)

Hakansv, 2018-02-11:
Pavel. I'm not sure I fully understand all your words but I succeed to update and fetch one Netherland inland chart by the chart downloader.

Hakansv, 2018-02-11:
Ok. Should I do more tests just tell me.

nohal, 2018-02-13:
I have packaged the new stuff and opened https://github.com/OpenCPN/OpenCPN/pull/991 to track the switch and tune the updates to the CI build. There is no documentation for the new build yet, but the only changes actually are the new VS version, dependencies and `v120_xp` -> `v141_xp` - https://github.com/nohal/OpenCPN/blob/wx31vs15_buildwin/appveyor.yml contains it all...

nohal, 2018-02-13:
@transmitterdan, we have merged the build system changes needed for wx3.1 and VS2017. Can you please rebase and submit the fixes (They should still be 3.0 compatible as we will almost certainly need to stick with that version on Linux for some more time )?
I have decided to take the dynamically linked `zlib` way as we need it on way too many places to link it statically all the time, otherwise it should be pretty straightforward.

transmitterdan, 2018-02-14:
Pavel,

I will start the rebase. I can now build master branch of O against wx3.1 latest git master.

Well done with the upgrade to the buildwin tools.

transmitterdan, 2018-02-14:
Pavel,

Please have a look at this rebase of the WX assert changes.

https://github.com/OpenCPN/OpenCPN/compare/master...transmitterdan:wx311

closed 2018-02-14 by nohal