OpenCPN/OpenCPN #861: OpenCPN does not close for some Windows 10 users

transmitterdan, 2017-07-27:
There are a few Windows 10 users (more as time goes by) that cannot get O to close cleanly. The reason is complex but this is a simplified explanation:

Source file chart1.cpp function void MyFrame::OnCloseWindow( wxCloseEvent& event ) calls wxYield() in 2 places to allow wxWidgets to continue processing events. There can be a lot of events piling up in the event queue if the computer is "slow". Especially so if there are a lot of NMEA messages arriving or leaving. There can be other external events that push events in to the event queue. I think there can be so many events the queue never gets empty and thus O hangs during exit.

Do we really need to flush the event queue during shutdown? Can we avoid call wxYield() during shutdown?

If we have to call wxYield() then it would seem we have to shut down all the incoming streams that generate events before we call wxYield().

bdbcat, 2017-07-27:
tdan...

The wxYield call is primarily used to get a last screen render showing the cross-hatching.  I suppose it could go away, but I prefer to leave it to indicate something is happening in the background.  Usually, the longest running thing we need to wait for is the raster chart compression thread pool to drain.  And we need user feedback for that process.

I suppose we could change the order of shutdown stuff so that the offending messages in the queue are processed first, and cleanly.  But it would be nice to know which messages get stuck, and why...

Dan, have you reproduced this?

Dave

transmitterdan, 2017-07-28:
Dave,

I cannot reproduce on my machines.

I think it is the NMEA data streams causing the problem. I am working with one of the users who can reproduce the problem. An initial test where I moved the shutdown of the NMEA connections before the wxYield() call has been successful. I will make a PR and you can look at it. I think it will not have any bad side effects.

Dan

douwefokkema, 2017-07-28:
The NMEA streams don't seem to cause the problem. When I disable all NMEA problem remains. When I disable OpenGL however, O closes down fine. Even when it is hanging on close down, I can still disable OpenGL, then shutdown follows immediately.

transmitterdan, 2017-07-30:
@douwefokkema 

Would you be kind enough to build a test branch of my fork of OpenCPN and tell me if it "fixes" your shutdown problem? My fork is:

https://github.com/transmitterdan/OpenCPN

The branch is "ShutdownFix". This is a fork of the present day master branch with an additional feature. It closes down the NMEA multiplexer earlier in the process. I am thinking that may be enough to get O to shutdown faster without removing the Yield() call.

douwefokkema, 2017-07-31:
Tested, but does not cure the problem. But O does close down fine now with all NMEA disabled from the start.  See attached opencpn.log which includes a sort description for each test. 

[opencpn.log.txt](https://github.com/OpenCPN/OpenCPN/files/1186806/opencpn.log.txt)

douwefokkema, 2017-08-03:
Solved in 4.6.22. Indeed removal of the wxYield() solved the issue.

closed 2017-08-04 by transmitterdan