OpenCPN/OpenCPN #866: Documentation and Garmin radars

fliot, 2017-08-03:
Speaking about Garmin radar support, in the documentation, it deals only with generic terms, no on exact known working models...

How could I know if such Radar will work with http://www.nauticom.fr/fr/radars-marine/garmin-gmr-18hd-3357.html as "Interface Type 2: OpenCPN Master Mode, no chartplotter" ???

Anybody knows ? Does a list of exact working models exist ?

Thanks in advance,

Regards.

rgleason, 2017-08-12:
Fiot, I think this is a topic/question for the forum. You're more likely to get a response. I belive Dave offered to make it work with garmin hd with aomeone else testing, and I think that was done, but this needs to be confirmed. Newer garmin radars have not been tested and it is unknown.

Also you are welcome to update the garmin docs after you learn more about the plugin. I can make you an editor.

fliot, 2017-08-14:
Thanks for the answer.

About your proposal to maintain such list,
I have github skills to maintain the documentation for sure, but I do not know at all the concepts on which the plugin have been developped (either internal and external principles/api/etc...), I'm certainly not the best person to maintain it.

About my research,
The forum shows the same lack of liable supported hardware list (and not only for Garmin devices).
As any end-user, thinking about OpenCPN as a global solution, before purchasing hardware, this is little bit a twilight zone...

Nevertheless, this opensource project is really interesting, mature and high level, and I thank all contributors to have participate to its success.

Regards.

rgleason, 2017-08-21:
fliot,
We depend on skilled and committed users to maintain the User Manual. You sound more like a programmer (having git skills) but we would be delighted to have a little rigor put into a succinct Hardware compatibility List!    

To create such a list you would need to be an editor (Pm me on Cruiser's forum  [rgleason]) or to create the list and attach it to a forum thread and ask me to include it. 

I would prefer to make you an editor so you can add what you know about the Garmin Plugin now!
So PM me!

Thanks.

nohal, 2018-01-15:
This type of questions certainly targets https://github.com/opencpn-radar-pi/radar_pi, not the main application.

closed 2018-01-15 by nohal