OpenCPN/OpenCPN #868: chart files with same filename in different directories not being displayed

LeaseOnLife, 2017-08-05:
Example

I have a couple files "200.kap" in different chart directories:

1. /DATA/BOAT&CRUISING/CHARTS/BSB/NGA/200.kap
2. /DATA/BOAT&CRUISING/CHARTS/BSB/WesternCarib/200.kap

I had added the second one today and created a chartgroup for for it. I was not able to get the WesternCarib/200.kap to display. The log shows

07:14:12 EDT: Searching directory: /DATA/BOAT&CRUISING/CHARTS/BSB/WesternCarib for *.kap
07:14:12 EDT: Loading chart data for /DATA/BOAT&CRUISING/CHARTS/BSB/WesternCarib/200.kap
07:14:12 EDT:    Retaining newer chart file of same name: /DATA/BOAT&CRUISING/CHARTS/BSB/WesternCarib/200.kap

The NGA 200.kap is "Rio de La Plata to Cape Horn", a very different part of the world than the second 200.kap, "Puerto de San Andres" on one of the Colombian islands. As a workaround I renamed the Western Carib file and it showed.

I think it is wrong to suppress chart files by file names. Specifically now that so many GE2KAP chartlets are floating around with random names.

As a secondary bug: the message in the log does not point to the colliding two filenames, it shows only the location of one of the two.

Dirk aka LeaseOnLife

transmitterdan, 2017-08-06:
Dirk,

What version of OpenCPN are you running and on what version operating system?

LeaseOnLife, 2017-08-06:
latest git on latest Artful Aardvark Ubuntu (upcoming 17.10)

transmitterdan, 2017-08-06:
Dirk,

Can you post the entire log file of the session?

LeaseOnLife, 2017-08-06:
I did a test  with a clean slate: _Folder-1 has the Colombian island chart, _Folder_2 has the NGA chart, both called 200.kap.

Started with an empty opencpn home directory, added _Folder-1 and _Folder-2 as chart directories. Result, opencpn olny shows the NGA 200.kap.  ==> opencpn.log.1.txt

closed opencpn, moved the log file, renamed the colombian chart, started opencpn, rescanned chart database, both charts are shown ==> opencpn.log.2.txt

[opencpn.log.1.txt](https://github.com/OpenCPN/OpenCPN/files/1203401/opencpn.log.1.txt)
[opencpn.log.2.txt](https://github.com/OpenCPN/OpenCPN/files/1203402/opencpn.log.2.txt)

I think this is sufficient evidence?

Dirk

transmitterdan, 2017-08-06:
Dirk,

I am not looking for evidence. I believe you. I am looking for log information because that helps me understand how the program got to the point of printing certain log messages. Without the entire log file I can't see what happened and in what order. No insult was intended.

transmitterdan, 2017-08-07:
Dirk,

Can you post your [ChartDirectories] section of opencpn.conf for the above test case?

LeaseOnLife, 2017-08-07:
Certainly!

[ChartDirectories]
ChartDir1=/DATA/BOAT&CRUISING/CHARTS/BSB/_Folder-1^4431882597477252185
ChartDir2=/DATA/BOAT&CRUISING/CHARTS/BSB/_Folder-2^11116360660470658547

do you have trouble reproducing? it looked pretty clear to me at first, but maybe stooopid things like a "&"  in the directory path get in the way?

bdbcat, 2017-08-07:
Guys...

Actually, this is intended behaviour.
Consider US NOAA Raster charts, distributed by region.  There are duplicates of the small scale charts here in the different downloadable sets.  We do not want to have these duplicates in the database, obviously, so we prune them on adding, selecting the most recent of the duplicates to keep.

I don't see any easy and foolproof way to identify different charts that share the same name.  We could parse the chart contents more fully, for instance by comparing the coverage area of identically named charts for general equality. But I fear that that would be too slow for the general case, so should only be done where there are name collisions.

This is evidently a rare, but real situation in the real world. Never been encountered before, AFAIK.
I'm open to other ideas that do not break existing behavior.

Dave

LeaseOnLife, 2017-08-07:
is O parsing the kap header at all, like for the edition, like where is O getting the info that one chart is newer than the other? If parsing already happens, it shouldn't be too costly to check one more field, like BSB/NA (name of the chart).

At a very minimum I would like to see more in the log file: the path of any file dropped and the file being used instead. I personally also would like to see a messages of " xx duplicate kap files found and only one instance for each duplicate will be displayed", but I understand we want to stand away from to many popup messages... 

I bet this is encountered more often than you think: People dock swap harddrive / USB sticks with large amounts of charts, they wouldn't know if a chart is dropped...

Cheers

Dirk

bdbcat, 2017-08-07:
Dirk...

All good points.  Should not be too hard to implement, just testing will be fussy....

Lets get this onto the Flyspray list as an error in 4.6+, and it will get addressed as resources allow.  I don't consider it to be a panic situation today.

Dave

transmitterdan, 2017-08-07:
Dirk,

If you create a Flyspray item I will add my comments there. I have an idea that I hope will make "most everyone" happy. And it is not too hard to implement as far as I can tell.

LeaseOnLife, 2017-08-08:
I will, give me a day, I am on a different computer and don't have my bugspray ID handy :-)

LeaseOnLife, 2017-08-08:
and for the testing part, I am happy to do it. Any additional testing scenarios, please let me know!

Dirk

LeaseOnLife, 2017-08-08:
moved to flyspray:
https://opencpn.org/flyspray/index.php?do=details&task_id=2261
therefor closing it here.

closed 2017-08-08 by LeaseOnLife