OpenCPN/OpenCPN #879: TCP listener causing segfault on connection

glambx, 2017-08-22:
Occasional crashes in the past, but now consistently crashes on TCP connection.

OpenCPN server configured to listen on TCP:0.0.0.0:10110 (output only).

Debian testing on client and server, kernel 4.11.0-2-amd64.  Same results with OpenCPN 4.4.0 (ie. not a regression).

Tried multiple different hosts (all amd64), purging ~/.opencpn and recreating, etc.  UDP seems to function normally.

OpenCPN (server side) is stable prior to inbound connection.  However, as soon as you telnet <server> 10110 or connect with another OpenCPN instance (in read mode), the UI locks up and jumps to 100% CPU utilization, and a few moments later crashes with with Abort, Segmentation fault, or a WxWidgets crash message.

Tried building packages locally with same results.  Can be reproduced simply by connecting to a listening output NMEA service via TCP with any Debian/testing or sid host.

Happy to provide any logs or assistance with debugging. :)

transmitterdan, 2017-08-23:
I ran latest O on Windows. Set up  connection TCP, address 0.0.0.0, port 2947. Then in a Cygwin window issued:

`$ telnet 127.0.0.1 2947`

This is the output:

```
Trying 127.0.0.1...
Connected to 127.0.0.1.
Escape character is '^]'.

!AIVDM,1,1,,A,4h3Ovk1uipCh`o>jOFEdjF?028GR,0*22

!AIVDM,1,1,,B,15MkCMPP0hG?q@BEa2?WEgw<08GS,0*51

$M1RMC,194839.00,A,3748.5904,N,12226.6216,W,7.9,353.7,160712,14.2,E,A*7C

!AIVDM,1,1,,B,4h3Ovk1uipCh`o>jOFEdjF?028GT,0*27

!AIVDM,1,1,,A,4h3OvkAuipCh`o?6BtEa1Rw02@GV,0*4E

!AIVDM,1,1,,B,4h3OvkAuipCh`o?6BtEa1Rw02@G`,`0*7B

!AIVDM,1,1,,A,4h3OvkQuipCh`o?Uh6Ef=VO020S:,0*6A

$M2HDM,340,M*44

$M1HDT,344.6,T*5B

!AIVDM,1,1,,B,4h3OvkQuipCh`o?Uh6Ef=VO020S:,0*69

$M2VHW,,T,340,M,08.18,N,15.14,K*1D

$M1HDG,330.4,,,14.2,E*6D

!AIVDO,1,1,,,B5MpaW00C5kou75J:>sKswl5wP06,0*16

$M2DBT,060.3,f,018.3,M,010.0,F*60

!AIVDM,1,1,,A,15Mw0LPPCho?T1<Ea5l5El?B2@Gi,0*06

$M2VWR,089,L,15.1,N,07.7,M,028.0,K*38

$M2MTW,014,C*47
```

transmitterdan, 2017-08-23:
No crash or wx warnings.

Will try in a Linux VM next.

glambx, 2017-08-23:
It _was_ working for me a few weeks ago (Debian 8.0), but seemed to break after upgrading my ship computer to Debian testing.  I'm wondering if it might be kernel / libc version related.

transmitterdan, 2017-08-23:
I am not sure what it might be. On my Ubunto 14.04 virtual machine I can get a connection but no output stream shows up on the telnet terminal. No crash or wx alerts. I will keep trying to troubleshoot.

O uses wx to do tha actual connecting. Maybe the old version 3.0 of wxWidgets doesn't work right with newer kernels. I will look into that as well.

nohal, 2018-01-15:
Still reproducible? If so, can we gat a backtrace from `gdb`?

nohal, 2018-02-04:
OK, I take it nobody is able to reproduce this.
Reopen if needed.

closed 2018-02-04 by nohal