OpenCPN/OpenCPN #884: Rollover windows do not time out when refresh is faster than 500ms

canboat, 2017-09-14:
The rollover window over AIS and routes is removed by a timer. However that timer is continuously reset to 500ms on every redraw of the chart. If there is a source that continuously asks for such redraws then the timer will never fire, and the rollover menu is stuck on screen.

This happens when the BR24radar_pi is told by the user to draw overlays onto the chart window and the refresh is faster than 2 x per second. Some radar users like this (in fact they like 10 redraws per second) so that the radar image looks like an old-fashioned radar and you see it 'sweeping' across the chart.

nohal, 2018-01-15:
@canboat https://github.com/OpenCPN/OpenCPN/commit/443579e496ed13820a78fb960c86279e45fecabf is merged so I suppose you are happy now, right?

canboat, 2018-01-15:
Yes, I am very happy!

closed 2018-01-15 by canboat