OpenCPN/OpenCPN #886: Plugins that draw on other window canvases cause screen corruption of chart window

canboat, 2017-09-14:
The OpenCPN source code does not call `wxGLCanvas::SetCurrent(const wxGLContext& context)` before all OpenGL calls.

The plugins were given the current `wxGLContext` until v4.4.0 when the switch to wxWidgets 3.0.0 was made where wxWidgets dropped the `GetCurrent()` method. Instead of passing a reference to the global variable they now get `NULL`. (See commit https://github.com/OpenCPN/OpenCPN/commit/3af7e9600ecfafb033f567d1e7f8667340924024 )

Unfortunately this puts a plugin that wants to use OpenGL on a different canvas in a hard place. It cannot leave out `SetCurrent()` completely, or the OpenGL graphics won't end up on the canvas. It also cannot call `SetCurrent(...)` with the OpenGL wxGLContext, as it is not told what that canvas is.

So for now the only option that it has is: create its own `wxGLContext`, call `SetCurrent()` on that context and then call `SetCurrent(NULL)` when it is done so that any OpenCPN OpenGL calls before OpenCPN does a `SetCurrent()` again are ignored. This strategy is largely successful, were it not for textures. What can happen is that textures are created or updated between the plugin `SetCurrent(NULL)` and the redraw happens after OpenCPN has called `SetCurrent()`
again. 

I thought that these cases were rare and corrected automatically, but it happens that AIS and route rollovers are drawn as described above, and when these are used corruption of the AIS rollover or even the entire chart can happen.

I see two strategies to fix this:
1. Make sure that `SetCurrent(m_pcontext)` is called everywhere in OpenCPN where OpenGL calls are made.
2. Pass the `m_pcontext` to the plugin in `RenderGLCanvas()`.

The latter would then depend on the plugin to either only use the (now shared) `wxGLContext` or use its own and restore the OpenCPN context when it is done with its own context. It is the least amount of work on OpenCPN's behalf though and it seems harmless enough.

I think the BR24radar_pi is the only plugin that uses 2ndary OpenGL canvases, and it has the necessary code for case 2 already.

I will create a pull request for case 2 (a single one word fix) shortly, but would be happy to discuss alternatives.

rgleason, 2017-09-14:
Will this break plugins?
Pass the m_pcontext to the plugin in RenderGLCanvas().

canboat, 2017-09-15:
> Will this break plugins?
Theoretically it is possible but now they are passed a NULL pointer so it is hard to see why they would break when they do get passed a non-null pointer, especially because that is how it worked before 4.4.0.

You could argue that since nothing broke in 4.4.0 then nobody was using it at the time?

I actually doubt that there are any besides BR24radar_pi that actually care at at all. For that they need to have a separate OpenGL window. Most plugins either don't use OpenGL at all (f.i. radar_pi draws AIS targets on different window using non-OpenGL calls) or draw on the chart window (f.i. wmm)

nohal, 2017-12-21:
Should be fixed by https://github.com/OpenCPN/OpenCPN/pull/887

closed 2017-12-21 by nohal