OpenCPN/OpenCPN #889: Degraded performance in recent releases

douwefokkema, 2017-09-15:
For the br24radar_pi I normally test also on a very slow machine: Intel Atom N$%) 1.66 Ghz, 2 GB, Win10.
Until release 4.5 this was running fine, with 4.8 it is extremely slow, and / or non-responding. But in this case also without the radar_pi. After a few zooms and or panning OpenCPN will hang indefinitely with a CPU load of 45%. Also without panning the static CPU is often above 20%.
I looked into this and a solution is in https://github.com/OpenCPN-Navico-Radar-Plugin/OpenCPN master
I will do a pull request for this.
Or may be simpler, remove the glEnable( GL_POLYGON_SMOOTH ); on line 9132 of s52plib.cpp
After this correction all is running as it used to be, also with radar window open.

seandepagnier, 2017-09-16:
I agree that GL_POLYGON_SMOOTH should be removed, but it should be replaced with proper multisampling to ensure antialiased polygons. 

Otherwise, maybe it can just be blacklisted for your specific graphics hardware so that we do not lose quality normally.

nohal, 2018-01-15:
Still relevant?

douwefokkema, 2018-01-16:
I have not tested latest versions of O for this issue. But I expect it to be still there. And I can't test now (out on holiday). Personally I am not really bothered by this issue. But there may be quite some people out there with older laptops that are. And normally they don't complain.
I use this slow laptop only for testing radar software under high load conditions, and I can easily compile a dedicated version of O for this testing. 
If you want O to remain compatible with older hardware it would be better to solve this, if compatibility with old hardware is not important, drop it.

nohal, 2018-02-04:
Should be addressed by #970, reopen if needed

closed 2018-02-04 by nohal