OpenCPN/OpenCPN #895: Snap package for OpenCPN is ready to test.

msis, 2017-09-28:
Hi,

I just finished making a `snap` for `OpenCPN`, please review it and give your comments.

Forum post: https://forum.snapcraft.io/t/call-for-testing-opencpn-4-8/2292
Source + Bug reports: https://github.com/msis/opencpn-snap

nohal, 2017-09-29:
Good work!
Seems like the time to take my Linux machines from 2014 to 2017 has finally come :)
For the snap I would probably consider including the tide data, the help docs and maybe some more detailed GSHHS basemap in it as I suppose for the end-user, adding these is a task far from trivial.
Much bigger it will become, of course. (see https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:developer_manual:developer_guide:modularized_packaging or https://github.com/OpenCPN/OpenCPN/pull/151 respectively)

davegardner, 2019-06-03:
I was unable to launch Thunderbird email client from Grib requester. Also, I had very strange issues accessing charts on Shared Folders (I ran it in a VBox Virtual Machine). The folder was available in Nautilus and via Terminal but I was told I had insufficient permission when trying to access the chart folder from OpenCPN.

closed 2021-04-13 by bdbcat