OpenCPN/OpenCPN #915: LNK2001	unresolved external symbol "int const wxEVT_DOWNLOAD_EVENT"

liulanghaitun, 2017-12-08:
when I compile opencpn 4.8.0 with  Microsoft Visual Studio 2013, I got the following error:

Error	LNK2001	unresolved external symbol "int const wxEVT_DOWNLOAD_EVENT" (?wxEVT_DOWNLOAD_EVENT@@3HB) D:\OpenCPN-master\build\plugins\chartdldr_pi\chartdldr_pi.obj	chartdldr_pi 

Who can help me to solve this problem?

nohal, 2017-12-08:
Did you clean the build directory before starting?
Did opencpn.lib build correctly before the plugin?

Never seen this (But also never built with VS2013, only with 2010 and 2015)

liulanghaitun, 2017-12-08:
Thank you for answering my question!
When I found this problem,I cleaned the compiling project.
I first build opencpn,and then compile whole project.but it had no effect.
I'll take your answer. 

thank you!

liulanghaitun, 2017-12-13:
when I  change to vs2015 version.The problem still exists.
final find this link:
[http://www.cruisersforum.com/forums/f134/opencpn-beta-test-4-1-925-release-153621-7.html](url)
[https://github.com/OpenCPN/OpenCPN/issues/773](url) ,the last error is same as me

the first link provided a patch,I solved this problem through the patch.

nohal, 2017-12-13:
The useless patch doing some weird stuff with `wxEVT_DOWNLOAD_EVENT` export (As you can see in that very thread, few posts down at http://www.cruisersforum.com/forums/f134/opencpn-beta-test-4-1-925-release-153621-7.html#post1928935 the "problem" the guy had does not exist and he was linking against static wx libraries instead of the dynamic ones), fixes the weird expat "problem", which I am also not seeing on a correctly configured system, doing the build without trying to be smarter than the instructions? I am getting a bit lost here...

transmitterdan, 2017-12-13:
I can compile latest git master on VS versions 2013, 2015 and 2017 with no errors. I can also link against wx V3.0.2, 3.0.3, 3.1 and latest wx git master dlls with no errors. So I any link error this be caused by a failure to follow directions.

nohal, 2017-12-13:
Well, if applying that bizarre patch "solved" something, he is very likely also erroneously linking against static WX. But as he provides no info it's still impossible to say for sure.

liulanghaitun, 2017-12-14:
Thank you for your reply .
This problem appeared when it was compiled into Lib.When I compiled it into DLL.The problem has been solved .

 thank you very munch

closed 2017-12-14 by nohal