OpenCPN/OpenCPN #922: PR: Improve track printing: Mix up of table headings

Hakansv, 2017-12-18:
@seandepagnier 
On Win10 built:VS2013
In commit "reimplement and improve track printing ability" (7f30599261) table headings "Distance" and "Course" (Or the data) are mixed up when both are selected.
[image]

[image]

nohal, 2018-01-08:
Hakan...
Can you still reproduce this? Seems to be fine here.

Hakansv, 2018-01-08:
Pavel..
Win10, today's build.
This is what I printed to a PDF just now

[image]

nohal, 2018-01-08:
Well, when you talk track and look at route... Of course I see the same...

commit e3058324 (2018-01-08): Correct column order in track printout table. Fixes #922
closed 2018-01-08 by nohal