OpenCPN/OpenCPN #957: Dashboard window properties reset after preferences modification

nohal, 2018-01-09:
After dashboard preferences are changed, the window is reset to floating/minimal size since https://github.com/OpenCPN/OpenCPN/pull/923

Hakansv, 2018-01-09:
It's the same when switching between horizontal and vertical. Neither user size is remembered in between. The dialog is restored to a min size, which is often to small for any instrument like the speedometer. Maybe because instrument now are scalable down to nearly invisible. So, if not a new dashboard the actual user size would be remembered in some way. That could be tricky if content is changed but keep at least the width?

Hakansv, 2018-01-09:
Sean.. another issue 
GPS status gauge seams not scalable:

[image]

bdbcat, 2018-01-11:
Mostly fixed by 52b43d8b912da86cddfcd0560f56da435e804b77

Thanks
Dave

rgleason, 2018-01-11:
I'd like to be very gentle about this, but  Dashboard is in recovery and on IV. She needs some help! and resuscitation.

bdbcat, 2018-01-11:
For example???

Hakansv, 2018-01-11:
Dave..
Commit  2a42a7113 "Increase Dashboard instrument minimum default size." wasn't the final solution for Windows at least. Some polishing may be needed:
- Combination depth graph and a speedometer-alike instrument has conflicts.  It's hard to describe, you better test yourself. Two pictures below shows I've to adapt for either instrument or graph, not both simultaneously. Maybe the depth graph can be better on variable seizing? 
[image] [image]

Before this commit it was possible to combine: 
[image]

Håkan

Hakansv, 2018-01-11:
Dave again: 
Sorry - I forgot to mention that horizontal/vertical switch and return after change of preferences is now much better since the gauges can't scrimp to a illegible size. 
It's sometimes to easy to focus on negatives instead of appreciation - or?
Thanks

bdbcat, 2018-01-11:
Hakan...
I partially reverted previous commit. 
I do not like the present logic for dial sizing, very fragile.  But nothing can be done for this Release.  I would truly love to see a re-write of all Dashboard layout logic.  It is a difficult problem to do automatically for all possible choices, all possible screen resolutions, etc..
And even a new, more modern look would be welcome.  Some day....

Thanks
Dave

Hakansv, 2018-01-11:
So, then we're back to this after changed preferences or horizontal/vertical switch, see pict.
Was there a good reason for the variable size commit at all?

[image]

OpenCPN, 2018-01-12:
Hakan...
Tweaked again.  Slightly better for Depth instrument.

The wxWidgets sizers do not allow us to do exactly what we want here, so solution is not perfect.
But good enough for this Release, I think.

Hakansv, 2018-01-12:
Thanks for the "Version 3.x xxDPT sentence" fix. I forgot to mention when the VDR file was sent. That fix was once a very old suggestion in Flyspray never included. :)

I can very well agree to "good enough" and its implications. 
The depth graph would though need a rescale in y-direction. When sized to MinWidth the bottom part, including the data curve, of the graph is not visible without manual resize of the Dashboard. 
This little tweak may solve it to "good enough". The graph scale is not longer accurate but visualization of an increasing depth is clear, and that would be the purpose of the graph? Depth readings are clearly labeled.
```
@@ -152,7 +152,7 @@ void DashboardInstrument_Depth::DrawForeground(wxGCDC* dc)
       dc->SetBrush(brush);
       dc->SetPen(*wxTRANSPARENT_PEN);
 
-      double ratioH = 100.0 / m_MaxDepth; // 140-40=100
+      double ratioH = 75.0 / m_MaxDepth; // 140-40=100
       double ratioW = double(size.x-6) / (DEPTH_RECORD_COUNT-1);
       wxPoint points[DEPTH_RECORD_COUNT+2];
```

bdbcat, 2018-01-12:
Hakan...
I implemented your suggestion.

Hakansv, 2018-01-12:
Thanks.
I've tested Dashboard on Win10, WinXP and RPi-Raspbian-Jessie with the little Raspberry 7 inch 800*400 150 mm screen.
All works fine.
Håkan

rgleason, 2021-04-23:
@Pavel or 2HakanSV  is this still useful or can it be closed?  Thanks.

Hakansv, 2021-04-23:
For me it's OK to close.
Håkan

bdbcat, 2021-04-29:
Closing, inactive

closed 2021-04-29 by bdbcat

pavel, 2021-06-06:
@rgleason Thanks for notifying me, but it does not seem that I'm involved in this project in any way.

https://blog.github.com/2011-03-23-mention-somebody-they-re-notified/

rgleason, 2021-06-06:
Sorry to have involved you but this is from a long time ago..