OpenCPN/OpenCPN #967: Implement direct download of the local grib sources

nohal, 2018-01-15:
To start with for [OpenSkiron](http://www.openskiron.org/en/)
[x] Get permission
[ ] Design the UI
[ ] Prototype

rgleason, 2021-09-10:
It would be wonderful if someone could provide a good PR for this long time rfe

tbultel, 2021-09-10:
From the history of this thread, I guess that there is no WIP on that feature.
If I am wrong, and if there has been some design thoughts please just tell me.
Else, from my point of view, since most of what we need is in xyGrib (list of servers,
connection status, list of available forecast models,
request based on a rectangle on the maps ...) it would sound reasonable to share code 
with them.  
Writing those lines I realized that grib_pi already contains code from the former zyGrib project.
The licensing should not be an issue, since OpenCpn already uses Gplv3 code.

nohal, 2021-09-10:
The licensing is not a problem, Qt vs. wxWidgets is though - we need to get rid of the Qt types while adopting any xyGrib code and reuse parts as a shared library, while not impossible, is somewhat complicated because the design did not anticipate it on either side.
And no, there is no progres on this since I discussed it with @norulz and got an OK for using the hi-res data he maintains on https://openskiron.org/en/

nohal, 2024-02-20:
Also https://www.euroszeilen.utwente.nl/weer/grib/

PR #3698 (2024-02-29): Grib plugin feature update
- Implements simple position prediction with set COG/SOG
- Implements download of specialized hi-res local models (So far Harmonie and the OpenSKIRON set around Europe)
  - Easily extensible as the data sources are stored in a catalog file that can be updated from the chart catalogs project maintained at https://github.com/chartcatalogs/gribcatalog/blob/main/sources.json 
- Implement dead-simple direct download of 0.4 degree ECMWF
  - I have written a small server counterpart in Go and shell and am currently hosting it on my private infrastructure

Closes #3697 
Clos [... cut]
closed 2024-02-29 by nohal