OpenCPN/OpenCPN #981: Refine the track simplification algorithm and GUI

nohal, 2018-01-28:
- [ ] The function is currently only available form the Route Manager right-click menu (= Hidden pretty well), make it more prominent (Track right-click menu, Route Manager button...)
- [ ] The current algo is OK for recorded tracks, but not good to simplify tracks exported from weather routing plugin routings. Simply adding bigger radiuses is not good enough - it can "eat" very significant parts of the routings towards turning points. (Try radius 1000.0 with the attached [Weather Route.gpx.txt](https://github.com/OpenCPN/OpenCPN/files/1671122/Weather.Route.gpx.txt))

Arne2024, 2026-01-17:
I stumbled on this old issue when looking into failure to use the track simplification in Weather Routing.
Track simplification does not work in WR-pi in current stable version 5.12.4. 

I suggest that when this is too hard to resolve to  remove the option from the WR plug-in, or deactivate, or label it as instable option in the WR plug-in.

nohal, 2026-01-17:
What does "does not work" mean exactly?

rgleason, 2026-01-18:
We aren't going to remove something in WR that works in the next version.

quinton-hoole, 2026-01-19:
> sorry for causing confusion, I had been testing weather routing scenarios over last few days. It seems to be working now, I can't replicate the message (I failed to save settings and the message) that along the line. It now saved a simplified route on all attempts tried out today. 
> so i will remove my comment. sorry again. 

No worries. That code is actually really messy. I wouldn't be at all surprised if there are some bugs in there. Don't be shy to report them if you find any.