OpenCPN/OpenCPN #1009: Decrementing iterator not allowed by MSVC V140

transmitterdan, 2018-02-28:
In void LLRegion::Optimize() at line 615 the iterator l is decremented. Using MSVC 2017 it throws a runtime exception that says such decrements are not permitted.

nohal, 2018-02-28:
What exactly does it complain? I am not aware of iterators not being allowed to be decremented in general. There are some cases where the behaviour is undefined though, like decrementing the iterator when it already is `.begin()`

transmitterdan, 2018-02-28:
I will send a screen shot of the runtime trap message. It is a show stopper on VS2017 debug mode. Debugger will not proceed beyond that point.

transmitterdan, 2018-03-01:
[image]

This is a screen shot of the trap.

transmitterdan, 2018-03-01:
It appears that the iterator is set to the beginning of the list. Then it is decremented which I believe is a genuine error.

transmitterdan, 2018-03-05:
I have made a branch in my fork that works in all platforms I can test (Windows, Linux, RPi). Let me know what you think.

https://github.com/OpenCPN/OpenCPN/compare/master...transmitterdan:LLOptimizeFix

rgleason, 2018-03-07:
https://github.com/OpenCPN/OpenCPN/pull/1014

transmitterdan, 2018-03-17:
This PR [#1005](https://github.com/OpenCPN/OpenCPN/pull/1005) seems to be the culprit. It will not run on modern Windows as all the runtime systems consider decrementing an iterator pointing to first element in the list as a hard error. I think the behavior is undefined as a minimum. I think this PR #1014 fixes it.

closed 2018-04-14 by transmitterdan