OpenCPN/OpenCPN #1011: Prepare a decent screenshot and host it on a fixed place on opencpn.org

nohal, 2018-02-28:
It would be good to have an official screenshot which actually is fully under our control and use it for stuff like #1010

leamas, 2018-02-28:
IWhile on it, make it two or three...

bdbcat, 2018-03-01:
OK, compose the shots, show me where they are, and we'll get them up on opencpn.org as publicly available.

nohal, 2018-03-01:
And of course, don't be shy and attach yours here if you think the world must see them ;)

leamas, 2018-03-01:
Will do, but  ENOTIME right now

rgleason, 2018-03-01:
Ok I'll bite, just to get this started.  Maybe I'll think of something clever to display O's capability, but simple things are usually better.
[image]
[image]

rgleason, 2018-03-01:
[image]

leamas, 2018-03-10:
[image]
[image]
[image]

FWIW, attaching my samples.

leamas, 2018-03-10:
Please note that att Appstream fomat has a strong bias to 16/9 images....

leamas, 2020-05-20:
@rgleason :
@hakansv:

Could you possibly make an update with fresh 5.0 screenshot(s), preferably two or three? We are about to  publish opencpn also on flathub, which uses the same appstream data as other linux distros like Debian/Ubuntu. The overall requirements for the images are in https://www.freedesktop.org/software/appstream/docs/chap-Quickstart.html, in short:'

  - png is preferred,  jpeg is also ok.
  - 16/9 aspect ratio.
  - At least 620 pixels wide.

rgleason, 2020-05-20:
800x450
[image]

1920x1080
[image]

leamas, 2020-05-20:
@rgleason : great, thanks!
@bdbcat : Could you upload to opencpn.org so we can get an "official" url (and eventually close this bug...)

bdbcat, 2020-05-20:
@leamas 
Upload the two pix?  Visible, or just as available download?  Must be on opencpn.org domain?

Dave

leamas, 2020-05-20:
@bdbcat :

>  Visible, or just as available download?

Just available for download is fine.

> Must be on opencpn.org domain?

I'm happy as long as you control it, be it on opencpn.org or somewhere else.

bdbcat, 2020-05-21:
Done.
https://opencpn.org/OpenCPN/assets/img/50info/82434780-611cbd00-9a61-11ea-8a30-bf44f098f875.png

https://opencpn.org/OpenCPN/assets/img/50info/82434783-624dea00-9a61-11ea-847b-347d91b8f0fd.png

leamas, 2020-05-21:
Great, thanks! 

However,  here is some problem (same for both urls):

```
$ wget https://opencpn.org/OpenCPN/assets/img/50info/82434783-624dea00-9a61-11ea-847b-347d91b8f0fd.png
--2020-05-21 04:52:19--  https://opencpn.org/OpenCPN/assets/img/50info/82434783-624dea00-9a61-11ea-847b-347d91b8f0fd.png
Resolving opencpn.org (opencpn.org)... 192.138.189.133
Connecting to opencpn.org (opencpn.org)|192.138.189.133|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2020-05-21 04:52:20 ERROR 403: Forbidden.

```

bdbcat, 2020-05-21:
Change of server...

http://download.opencpn.org/screenshots/82434783-624dea00-9a61-11ea-847b-347d91b8f0fd.png

http://download.opencpn.org/screenshots/82434780-611cbd00-9a61-11ea-8a30-bf44f098f875.png

leamas, 2020-05-21:
@bdbcat : Much better server ;)

@nohal: Seems that we have "an official screenshot which actually is fully under our control". I suggest that this bug is closed.

leamas, 2020-10-18:
@nohal: @bdbcat:  time to close?

closed 2020-10-19 by bdbcat