OpenCPN/OpenCPN #1016: Build Failure: error: using typedef-name ‘sqlite3_value’ after ‘struct’

ph1l, 2018-03-15:
```
~/code/OpenCPN$ git log --oneline | head -1
94e05c6a Fix the Japanese, Arabic and Vietnamese localizations build and packaging.
~/code/OpenCPN$ mkdir build; cd build; cmake -DBUNDLE_TCDATA=ON -DBUNDLE_GSHHS=FULL -DBUNDLE_DOCS=ON .. 2>&1 | tee /tmp/build_output.log
~/code/OpenCPN/build$ make 2>&1 | tee -a /tmp/build_output.log
SNIP
[ 15%] Building CXX object CMakeFiles/SQLiteCpp.dir/src/SQLiteCpp/src/Backup.cpp.o
/usr/bin/c++  -DLINUX_CRASHRPT -DOCPN_HAVE_X11 -DOCPN_USE_PORTAUDIO -DSQLITE_USE_LEGACY_STRUCT -DTIXML_USE_STL -DUSE_GARMINHOST -DUSE_LZMA -DUSE_RENDER_CAIRO -DUSE_S57 -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D__OCPN_USE_CURL__ -D__WXGTK__ -DocpnUSE_GL -DocpnUSE_SVG -DwxsvgUSE_EXIF -I/home/elektron/code/OpenCPN/src/mygdal -isystem /usr/lib/x86_64-linux-gnu/wx/include/gtk2-unicode-3.0 -isystem /usr/include/wx-3.0 -I/home/elektron/code/OpenCPN/include -I/home/elektron/code/OpenCPN/src -I/usr/include/x86_64-linux-gnu -I/home/elektron/code/OpenCPN/src/wxcurl -I/usr/include/cairo -I/home/elektron/code/OpenCPN/src/wxsvg/include/wxSVG -I/home/elektron/code/OpenCPN/src/wxsvg/include/wxSVGXML -I/home/elektron/code/OpenCPN/src/wxsvg/include -I/home/elektron/code/OpenCPN/src/wxsvg/src -I/usr/include/freetype2 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/atk-1.0 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/pango-1.0 -I/usr/include/gtk-2.0 -I/usr/lib/x86_64-linux-gnu/gtk-2.0/include -I/home/elektron/code/OpenCPN/src/SQLiteCpp/include -I/home/elektron/code/OpenCPN/src/nmea0183 -I/home/elektron/code/OpenCPN/src/garmin/jeeps -I/home/elektron/code/OpenCPN/src/texcmp/lz4 -I/home/elektron/code/OpenCPN/src/texcmp/squish -I/home/elektron/code/OpenCPN/build/CMakeFiles/include  -pthread -O3 -DNDEBUG   -Wall -Wno-unused -fexceptions -rdynamic  -g -fno-strict-aliasing  -DPREFIX=\"/usr/local\" -std=gnu++11 -o CMakeFiles/SQLiteCpp.dir/src/SQLiteCpp/src/Backup.cpp.o -c /home/elektron/code/OpenCPN/src/SQLiteCpp/src/Backup.cpp
In file included from /home/elektron/code/OpenCPN/src/SQLiteCpp/src/Backup.cpp:16:0:
/usr/include/sqlite3.h:3847:16: error: using typedef-name ‘sqlite3_value’ after ‘struct’
 typedef struct sqlite3_value sqlite3_value;
                ^~~~~~~~~~~~~
In file included from /home/elektron/code/OpenCPN/src/SQLiteCpp/include/SQLiteCpp/Backup.h:14:0,
                 from /home/elektron/code/OpenCPN/src/SQLiteCpp/src/Backup.cpp:12:
/home/elektron/code/OpenCPN/src/SQLiteCpp/include/SQLiteCpp/Database.h:27:20: note: ‘sqlite3_value’ has a previous declaration here
 typedef struct Mem sqlite3_value;
                    ^~~~~~~~~~~~~
In file included from /home/elektron/code/OpenCPN/src/SQLiteCpp/src/Backup.cpp:16:0:
/usr/include/sqlite3.h:3847:30: error: conflicting declaration ‘typedef int sqlite3_value’
 typedef struct sqlite3_value sqlite3_value;
                              ^~~~~~~~~~~~~
In file included from /home/elektron/code/OpenCPN/src/SQLiteCpp/include/SQLiteCpp/Backup.h:14:0,
                 from /home/elektron/code/OpenCPN/src/SQLiteCpp/src/Backup.cpp:12:
/home/elektron/code/OpenCPN/src/SQLiteCpp/include/SQLiteCpp/Database.h:27:20: note: previous declaration as ‘typedef struct Mem sqlite3_value’
 typedef struct Mem sqlite3_value;
                    ^~~~~~~~~~~~~
make[2]: *** [CMakeFiles/SQLiteCpp.dir/build.make:66: CMakeFiles/SQLiteCpp.dir/src/SQLiteCpp/src/Backup.cpp.o] Error 1
make[2]: Leaving directory '/home/elektron/code/OpenCPN/build'
make[1]: *** [CMakeFiles/Makefile2:192: CMakeFiles/SQLiteCpp.dir/all] Error 2
make[1]: Leaving directory '/home/elektron/code/OpenCPN/build'
make: *** [Makefile:155: all] Error 2
```

Possibly Related: https://github.com/OpenCPN/OpenCPN/pull/1008/files

Build Log: 
[build_output.log](https://github.com/OpenCPN/OpenCPN/files/1817280/build_output.log)

ph1l, 2018-03-15:
> libsqlite3-dev is already the newest version (3.22.0-1).

nohal, 2018-03-15:
On what are you trying to build it? The CI builds are passing on both Precise and Xenial and my local test on Bionic-pre as well.

ph1l, 2018-03-15:
debian sid

transmitterdan, 2018-03-15:
On Windows it builds just fine with either the Windows dependencies or the sqlite3 package from vcpkg. So must be something peculiar to your installation.

nohal, 2018-03-15:
It seems SQLiteCPP actually gets built before sqlite3.h is included in mbtiles.cpp. What I do not understand is why. I would like to avoid the clear solution of including it in Database.h, which upstream tries to avoid so hard by all these defines...

ph1l, 2018-03-15:
That's prolly not the right way to fix this, but the compile continues, this way...

PR #1017 (2018-03-16): Reorder includes so Database.h can tell if the define is needed
fixes #1016
closed 2018-03-16 by nohal