OpenCPN/OpenCPN #1022: Option "-c" for start OpenCPN with another configdir

free-x, 2018-03-19:
I would like to have the option "-c" for start opencpn with other configuration directory than default . e.g. "opencpn -c ~/.opencpn2".
Then it is possible use another instance of opencpn on the same computer 

Regards
free-x

Hakansv, 2018-03-19:
@free-x 
This already available and widely used by the -p option like, in Windows, opencpn.exe -p 
See: https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:advanced_features:portable_opencpn

free-x, 2018-03-19:
@Hakansv 
I'm Linux user :-) and options "-p" doesn't work for me 
also for windows users this could be a good alternative instead of cloning all binaries into another directory

nohal, 2018-03-19:
My crystal ball envisions a nightmare with plugins. But would be handy also for flatpak/snap...

free-x, 2018-03-19:
@nohal 

Don't understand, where is a problem....
First instance starts with default configuration from ~/.opencpn/opencpn.conf and another Instance for example from ~/.opencpn2/opencpn.conf.

nohal, 2018-03-19:
And then we have 20+ plugins that are not aware such a situation may exist to fix... OpenCPN ecosystem is not just the core application and 1 configuration file.

bdbcat, 2018-03-19:
It might "just work" if the plugins religiously use the API method:

extern "C"  DECL_EXP wxFileConfig *GetOCPNConfigObject(void);

But do they?  Mine do...

nohal, 2018-03-19:
For the config object it will, sure for most. But will it for data files? I am quite sure it won't.

bdbcat, 2018-03-19:
Well, yeah, didn't think of that.  My plugins would definitely corrupt themselves.

transmitterdan, 2018-03-20:
If portable mode does not work on Linux maybe we should fix that. Portable mode exists to allow running multiple copies of O with each it's own configuration (and data). Charts can be shared or not.

free-x, 2018-03-21:
Okay, all clear first. Portable also runs on Linux. This works like this
```
$ mkdir portable_o
$ cd portable_o
$ cp /usr/bin/opencpn .
$ cp -a /usr/share/opencpn/* .
$ cp /usr/lib/opencpn/* plugins/

./opencpn -p
```
Proof in the attachment
[image]

Do you still want to pursue the topic?

nohal, 2018-03-21:
I still think this has some merit, at least towards snap/flatpak...

RooieDirk, 2018-04-18:
Wouldn't it then also be nice to make it possible to change configuration while running O. Then you could easy have different configurations for say, deep sea / coastal / approach, or skipper and mate have there own settings.
In the big ECDIS world it is compulsory to have a way to switch to a known setting. So if you are come on watch, you can easy change all settings to a known (to you) value. 
This would prevent for ea, you coming on watch and thinking there in no other shipping around, and then discover later that the AIS was switched off.

leamas, 2019-04-10:
We have a new situation after 5.0.0. As @nohal says, this is required for flatpak. More exactly, any plugin using plugin data needs to use the new API function GetPluginDataDir() to locate the data.

Also, even if the details are far from clear, I'm convinced that this is also related to #1343 and #1347. 

TL; DR We will have to patch plugins to use the "official" ways in the API to locate their data in any case.

leamas, 2019-05-05:
For linux and according to the freedesktop specs (https://specifications.freedesktop.org/basedir-spec/basedir-spec-0.6.html) the default configdir should actually be `~/.config/opencpn` rather than `~/.opencpn`. Applications shoiuld also respect XDG_CONFIG_DIR which, if present, defines an alternative to `~/.config`.

This not a major problem right now, but should IMHO be taken in account when implementering a fix for this very issue.

leamas, 2019-05-05:
@bdbcat  wrote:

> It might "just work" if the plugins religiously use the API method:

>      extern "C" DECL_EXP wxFileConfig *GetOCPNConfigObject(void);

>  But do they? Mine do...

Perhaps not all do. But if so, this is IMHO bugs that should be fixed. With 5.0.0 out of the doors this is perhaps the time to update this function to make it possible to change the directory, and let the failing plugins break. After all, this is not a hard break, it will just happen if users actually invoke the new -c option.

Keeping compatibility is certainly a good thing. But to stay compatible with plain bugs in the plugins is perhaps to constrain development too much.

Someone should make a PR.

leamas, 2021-04-13:
After 5.2.0 and the new plugin installer most plugins out of necessity uses sane functions for getting the data and config directory.  As I understand it, implementing support for XDG_CONFIG_DIR should now be pretty straight-forward in linux. A -c option could be added as syntactic sugar, basically doing the same thing.

leamas, 2021-05-06:
This is related to #2256. Basically, plugins which writes data need to use GetpPrivateApplicationDataLocation() to get a reference to the directory where files should be written.  Some plugins don't,  this is bugs which must be fixed for many reasons (flatpak is just one of them).

Until this is sorted out it's probably no point pushing this -c issue further.

nohal, 2024-02-17:
Time to revisit...

leamas, 2024-03-03:
Tomorrow...

PR #3719 (2024-03-05): Add a new -c --configdir command line switch
Add a new command line option which uses an alternative configuration directory instead of the linux default ~/.opencn

Closes: #1022
closed 2024-03-05 by leamas