OpenCPN/OpenCPN #1034: MBTiles transparency support

willemvanopstal, 2018-04-05:
I converted some digital charts to the MBtile format to serve as an extra overlay on my navigational (ENC) charts.

While the MBTiles are based on RGBA colorvalues (with a large portion being ```NoData > A=0```), openCPN won't display this as a transparent layer.

Is transparency currently supported at all, or does it have a bug somewhere? Also, is there any specification available for building the MBTiles for use in openCPN?

willemvanopstal, 2018-04-05:
I have prepared two example files generated with Safe FME, writing in the SQLite MBTile format. You can find them [here](https://github.com/willemvanopstal/nautin_tiles/tree/master/mbtile_experiments). One file is generated with palettes and the other with embedded bands, both contain an alpha channel but are treated differently in openCPN. The paletted version is just neglecting the alpha-channel (seems to do) and the bands-version is changing representation to black.

Furthermore, the ```name``` is not accepted/displayed in openCPN metadata, while GDAL recognizes it correctly.

nohal, 2018-04-05:
Transparency is not currently supported at all.

The metadata available in the mbtiles datasets seem to vary a lot from none (most of the time), over wrong, to "correct" as in your case.

Showing the name if available in the metadata is simple, just not implemented at the moment, we will sure get to it when enough samples are collected to decide how to approach the metadata. I think we should support what NOAA provides in their datasets, which is a superset of what you use
```
"profile"	"mercator"
"attribution"	"NOAA"
"name"	"MBTILES_08"
"format"	"png"
"basename"	"MBTILES_08"
"bounds"	"-225, 47.0401821448, -191.25, 11.1784018737"
"minzoom"	"3"
"version"	"1"
"template"	"{{#__teaser__}} <div class='tooltip'> <strong>{{name}} {{kap}} </strong> <small class='tooltip'> NTM Date: {{ntm_date}} Scale: {{scale}} Units: {{units}} {{path}} </small> </div> <style type='text/css'> .tooltip strong { display:block; font-size:16px; } .tooltip small { font-size:10px; display:block; overflow:hidden; max-height:90px; line-height:15px; }</style> {{/__teaser__}}"
"maxzoom"	"19"
"tilejson"	"2.0.0"
"scheme"	"xyz"
"type"	"overlay"
"description"	"PACIFIC TERRITORIES"
```

willemvanopstal, 2018-04-05:
That's a pity, but thanks for the heads-up.
I personally think OpenCPN can really benefit from transparent layers like these, any thoughts on developing support for transparency?

btw. This [resource](http://docs.safe.com/fme/html/FME_Desktop_Documentation/FME_ReadersWriters/mbtiles/mbtiles.htm) may be useful for further development, just to see what's possible in different tile-generating software.

nohal, 2018-04-05:
The MBTiles implementation OpenCPN is an unfinished feature being currently worked on, keep that on mind while using and judging it..

willemvanopstal, 2018-04-05:
Sorry, don't want to judge anything here. I actually want to help and am looking into the new possibilities available, maybe give some ideas...

bc38, 2018-04-07:
Transparency is great!
It will be possible to make overlays on ENC
Similar implemented in MaxSea TimeZero
For the support of the Alpha channel!

ufslacker, 2026-11-06:
Proposed implementation for whole-tileset MBTiles overlay opacity (shader uniform + Options slider + `MBTilesOpacity` config): https://github.com/OpenCPN/OpenCPN/pull/5515

bc38, 2026-11-06:
It would be far more functional to overlay ENC point features and depth contours onto the raster - much like how it’s implemented in modern charts on chartplotters.