OpenCPN/OpenCPN #1040: chart text national bugs

crazysong, 2018-04-09:
in s52plib.cpp

S52_TextC *s52plib::S52_PL_parseTE( ObjRazRules *rzRules, Rules *rules, char *cmd )
// same as S52_PL_parseTX put parse 'C' format first
{
...
       // get value for this attribute
       **parg = _getParamVal( rzRules, parg, val, MAXL );**

it doesn't show national text,
so ,  i think it can be modified to :

=============================

                // get value for this attribute
                if (!strcmp("OBJNAM", arg) && ps52plib->m_bShowNationalTexts) {
                    // in case user wants the national text shown and the rule contains OBJNAM, try to get the value
                    parg = { "NOBJNM" };
                    parg = _getParamVal(rzRules, parg, val, MAXL);
                    if (!strcmp("NOBJNM", val)) {
                        parg = { "OBJNAM" };
                        parg = _getParamVal(rzRules, parg, val, MAXL); //NOBJNM is not defined
                    }
                    else {
                        val[MAXL - 1] = '\0'; // make sure the string terminates
                    }
                }
                else {
                    parg = _getParamVal(rzRules, parg, val, MAXL);
                }
===============================================

**do you think it's ok??**

did-g, 2018-04-23:
Hi,

Do you know a chart with parseTE and OBJNAM? You're opening a big can of worms here  :)

Something like your code should work but all this stuff (s52lib and co) is frightening.

Regards
Didier

bdbcat, 2021-04-12:
Inactive

closed 2021-04-12 by bdbcat