OpenCPN/OpenCPN #1043: double wxCurl.... loading warning/error

RooieDirk, 2018-04-12:
Since a few weeks I get a warning/error at startup. Compiling is fine.
Running linux Mint.
The info wxwidgets is throwing up.

ASSERT INFO:
../src/common/object.cpp(251): assert "classTable->Get(m_className) == NULL" failed in Register(): Class "wxCurlDownloadEvent" already in RTTI table - have you used IMPLEMENT_DYNAMIC_CLASS() multiple times or linked some object file twice)?

BACKTRACE:
[1] wxClassInfo::Register()
[2] dlopen
[3] wxDynamicLibrary::RawLoad(wxString const&, int)
[4] wxDynamicLibrary::Load(wxString const&, int)
[5] wxDynamicLibrary::wxDynamicLibrary(wxString const&, int) /usr/include/wx-3.0/wx/dynlib.h:241
[6] PlugInManager::LoadPlugIn(wxString) /home/dirk/Development/OpenCPN_Dirk/OpenCPN/src/pluginmanager.cpp:1175
[7] PlugInManager::LoadAllPlugIns(wxString const&, bool, bool) /home/dirk/Development/OpenCPN_Dirk/OpenCPN/src/pluginmanager.cpp:413
[8] MyFrame::OnInitTimer(wxTimerEvent&) /home/dirk/Development/OpenCPN_Dirk/OpenCPN/src/chart1.cpp:6614
[9] wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const
[10] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[11] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[12] wxEvtHandler::TryHereOnly(wxEvent&)
[13] wxEvtHandler::ProcessEventLocally(wxEvent&)
[14] wxEvtHandler::ProcessEvent(wxEvent&)
[15] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[16] wxTimerImpl::SendEvent()
[17] g_main_context_dispatch
[18] g_main_loop_run
[19] gtk_main
[20] wxGUIEventLoop::DoRun()
[21] wxEventLoopBase::Run()
[22] wxAppConsoleBase::MainLoop()
[23] wxEntry(int&, wchar_t**)
[24] main /home/dirk/Development/OpenCPN_Dirk/OpenCPN/src/chart1.cpp:921
[25] __libc_start_main
[26] _start

RooieDirk, 2018-04-12:
Kees,
Stupid. It is indeed a plugin As soon as I found the offending PI, I'll let it know.

did-g, 2018-04-18:
It's likely oesenc_pi or ofc_pi and it's because you're compiling them inside opencpn tree, there's a namespace  pollution between O and plugins, you have to compile them outside O tree or apply a patch like
```
commit 87e7b6c8ea62a0502e09c6f12898db006b003cd2

    use a private name for WXCURL source list or it may use SRC_WXCURL list from OpenCPN tree

diff --git a/CMakeLists.txt b/CMakeLists.txt
index b1ee9d0..6f42649 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -205,7 +205,7 @@ IF(NOT QT_ANDROID)
 
 
     IF(WIN32)
-        SET(SRC_WXCURL
+        SET(SRC_OFC_WXCURL
         src/wxcurl/base.cpp
         src/wxcurl/dav.cpp
         src/wxcurl/davtool.cpp
@@ -252,7 +252,7 @@ ENDIF(TINYXML_FOUND)
 ADD_DEFINITIONS(-DTIXML_USE_STL)
 
 
-ADD_LIBRARY(${PACKAGE_NAME} SHARED ${SRC_OFC} ${SRC_WXCURL} ${SRC_LTINYXML})
+ADD_LIBRARY(${PACKAGE_NAME} SHARED ${SRC_OFC} ${SRC_OFC_WXCURL} ${SRC_LTINYXML})
 
 IF (QT_ANDROID )
     TARGET_LINK_LIBRARIES( ${PACKAGE_NAME} ${OCPN_Core_LIBRARIES} )

```