OpenCPN/OpenCPN #1062: Proper types for coordinates (lat+lon) and headings

taruti, 2018-04-28:
OpenCPN is using double in lots of places and doing informal operations on heading and latitudes/longitudes that are wrong in edge cases. I am thinking of introducing datatypes
(class LatLon and class Heading) and defining all needed operations on them. 

This should not make things slower or take up more memory, but it will fix many bugs around edge cases.

Would this be acceptable in principle? I have already around some of the code and the change can be made gradually and the new solution can co-exist with the old one. Is there a better place to chat about this?

rgleason, 2018-04-28:
Taruti would this also hopefully lead to/help make the "Right Click" copy and paste of Lat/Long between Opencpn and Plugins a seamless operation?   Currently there are lots of format issues, we need to settle on a single format for copy/paste while having the conversion to what the user has selected is done automatically in the background.  I think we should be using decimal degrees for copy/paste.  

We also have a Safety and Rescue SAR plugin that has been withdrawn because user's may enter the wrong format of lat/long and thus will create SAR routes that are incorrect.  [FS#2110 - Conversion error. Dangerous result! ](https://opencpn.org/flyspray/index.php?do=details&task_id=2110)

A consistent lat/long copy/paste format in opencpn and all plugins would help to fix that problem.
See [FS#2342 - API 1.15 Lat/Long Format for waypoint copy/paste Interoperability ](https://opencpn.org/flyspray/index.php?do=details&task_id=2342)

This Lat/Long interoperability issue ripples through many parts of the code and touchs parts of Opencpn and most all plugiins. It would be great to get this resolved effectively, but it may be a multistep process starting with Opencpn.

I am hopeful that there will also be some example code for all plugins to use so that they can take advantage of this too. 

I believe that the Plugin API might have to change, but if everything used a consistent format it might not be necessary.  See this suggestion to Jon Gough too (made to hime since he has authored a number of plugin API additions).   [API 1.15 Extension ? #382](https://github.com/jongough/ocpn_draw_pi/issues/382)

I think it would be great if a capable programer could take this issue on, but I am not Dave!

nohal, 2018-04-29:
This place is quite fine.

Yes, and we should do something about the legacy of global vars holding the information about the own ship navigational info and settings in general, probably creating a class encapsulating this info - will come handy also for the upcoming SignalK implementation which should if not replace completely at least avoid the current NMEA0183 workflows spread over half a gazillion places in the application.

Just try to break the refactoring into manageable pieces and send PRs for them "early" as we have already seen a bitter ending attempt to refactor everything at once - Does not work.

leamas, 2018-08-25:
Just a note: The upstream NMEA library at https://github.com/SammyB428/NMEA0183 i. e. the updated version of our bundled src/nmea0183 seems to have this partly in place. Might be  a natural starting point.

rgleason, 2019-02-01:
Have added a comment / summary to 
[FS#2342 - API 1.15 Lat/Long Format for waypoint copy/paste Interoperability ](https://opencpn.org/flyspray/index.php?do=details&task_id=2342)

RooieDirk, 2019-07-02:
Qt has since 5.2 a nice class for this that would almost be perfect for us I guess (Only transfer to wxWidgets): QGeoCoordinate Class
https://doc-snapshots.qt.io/qt5-5.11/qgeocoordinate.html

RooieDirk, 2019-08-25:
How about something like this?? Not complete yet;-)

`
class OCPN_Angle
{
public:
    OCPN_Angle();
    inline OCPN_Angle (double a){angle = a;};
    ~OCPN_Angle();
    SetDeg( double a){angle = a;};
    SetRad( double a){angle = a * RADIAN};
    double GetDeg(){return angle;};
    double GetRad(){return angle * DEGREE;};
    wxString GetString( );
protected:
    double angle;
};

class OCPN_Latitude 
{
public:
    OCPN_Latitude();
    OCPN_Latitude( OCPN_Latitude l){lat.SetDeg( l.GetDeg());};
    ~OCPN_Latitude();
    void SetDeg(double l){ lat.SetDeg( ReduceLat( l ) );};
    void SetRad(double l){ lat.SetRad( ReduceLat( l ) );};
    double GetDeg(){return lat.GetDeg();};
    double GetRad(){return lat.GetRad();};
    double GetExaggerated(){ return 10800/PI * log( tan( PI/4 + lat.GetRad())/2 ));};
    double GetDeltaDeg( OCPN_Latitude l2){return l2.lat.GetDeg()-lat.GetDeg();};
    double GetDeltaRad( OCPN_Latitude l2){return l2.lat.GetRad()-lat.GetRad();};
    wxString GetString( );
protected:
    double ReduceLat( double l){if (l<-90) l=-90; if(l>90) l=90; return l;};
    OCPN_Angle lat;
    
};

class OCPN_Longitude 
{
public:
    OCPN_Longitude();
    OCPN_Longitude( OCPN_Longitude l){lon.SetDeg( l.GetDeg());};
    ~OCPN_Latitude();
    void SetDeg(double l){ lon.SetDeg( ReduceLat( l ) );};
    void SetRad(double l){ lon.SetRad( ReduceLat( l ) );};
    double GetDeg(){return lon.GetDeg();};
    double GetRad(){return lon.GetRad();};
    double GetDeltaDeg( OCPN_Latitude l2){return l2.lon.GetDeg()-lon.GetDeg();};
    double GetDeltaRad( OCPN_Latitude l2){return l2.lon.GetRad()-lon.GetRad();};
    wxString GetString( );
protected:
    double ReduceLon( double l){if (l<-180) l=-180; if(l>180) l=180; return l;};
    OCPN_Angle lon;    
};

class OCPN_Position
{
public:
    OCPN_Position();
    OCPN_Position(double la, double lo);
    OCPN_Position(OCPN_Position pos);
    ~OCPN_Position();
    void SetLatDeg(double l){ lon.SetDeg( l );};
    void SetLatRad(double l){ lon.SetRad( l );};
    void SetLonDeg(double l){ lon.SetDeg( l );};
    void SetLonRad(double l){ lon.SetRad( l );};
    double GetDistMerc( OCPN_Position p);
    double GetBearMerc( OCPN_Position p);
    double GetDistGcSphere( OCPN_Position p);
    double GetBearGcSphere( OCPN_Position p);
    double GetDistGcEllip( OCPN_Position p);
    double GetBearGcEllip( OCPN_Position p);
    OCPN_Position GetNewPos(OCPN_Angle a, double d );
    wxString GetString( );    
protected:
    OCPN_Latitude lat;
    OCPN_Longitude lon;
};

class OCPN_Obj
{
public:
    OCPN_Obj();
    OCPN_Obj( OCPN_Obj o);
    ~OCPN_Obj();
    void SetPos( OCPN_Position p);
    void SetCourse( OCPN_Angle a);
    void SetSpeed( double s);
    void SetTime( wxDateTime);
    OCPN_Position GetPos();
    OCPN_Angle GetCourse( );
    double GetSpeed();
    wxDateTime GetTime();
    OCPN_Position GetCrossPos(OCPN_Obj o);
    double GetCPA(OCPN_Obj o);
    wxDateTime GetTCPA(OCPN_Obj o);
    double GetBCR(OCPN_Obj o); //Bow Crossing Range
    wxDateTime GetTBCR(OCPN_Obj o);
    OCPN_Angle GetRelativeCourse(OCPN_Obj o);
    double GetRelativeSpeed(OCPN_Obj o);    
protected:
    OCPN_Position pos;
    OCPN_Angle course;
    double speed;
    wxDateTime time;
};
`

taruti, 2019-08-27:
Had some code for this lying around, but I think that this is not a good moment to bring it in. We have many open PRs and the android code is lagging behind. And this would conflict with lots of those...

RooieDirk, 2019-08-28:
I'm afraid the 'right' moment will never arrive. As I see it we could just add a file with code and slowly start using it in the existing code. 
Just adding a file will not break anything but does give an opportunity to start .
Anyhow I would like to see what your ideas are. See if we can make something well thought out, and that goes better with more minds involved.

taruti, 2019-08-29:
Ok, I'll post stuff next week. (need to get some work stuff + sailing done).