OpenCPN/OpenCPN #1065: Question: how compliant is the s52 rendering w/ IHO specs?

jaredrada, 2018-05-02:
I am hoping to find out how closely openCPN follows the S52 publication guidelines for ECDIS? I have skimmed over S52 and Annex A (v4.0) which appear to have very verbose/lengthy guidelines on rendering methodology. How closely does openCPN follow these guidelines? I have looked at some of the src files and notice the s52 presentation lib is substantial. 

I am interested in writing my own navigation/ENC rendering for academic purposes so researching how some freely available applications handle this aspect. I am not looking to rewrite 10-20k lines of C++ code to achieve my goals but may have to reduce expectations or leverage available code.

Thanks for any insight.

nohal, 2018-05-02:
As closely as possible with the information available. It is implementing an older version of the standard (not available anywhere for public download at the moment as far as I can tell) and only a very small subset of 4.0 related to display of isolated dangers bellow safe depth contour was implemented to it so far (There sure are changes in the CS procedures and for example sounding symbols were replaced with vector versions in 4.0).
If you manage to get hold of the ECDIS Chart nr. 1 and the reference renderings (as far as I know it is part of the $500 NDA bound S52 presentation library package from IHO), you can probably compare for yourself.

jaredrada, 2018-05-03:
Thanks for the info!

closed 2018-05-03 by jaredrada