OpenCPN/OpenCPN #1069: Multi thread OCPN for multi processor machines

jongough, 2018-05-03:
Current CPU's seem to be going for more cores rather than increasing the speed of the cores. Raspberry Pi is an example of 4 cores that are not overly fast. Currently OCPN uses, or appears to use, a single thread of execution for everything. With the addition of more functionality and the use of more plugins the processing that OCPN does overloads slower cpu's. Low powered PC's can be overwhelmed and the screen updates will slow down and make the user interaction very difficult.

A solution would be to make OCPN multithreaded such that plugins, screen updates, messaging, etc. run in separate threads which could then be scheduled on the multiple cores available. Currently only eventing is running on separate threads. This would make low power machines, like the Raspberry PI, much more usable.

keesverruijt, 2018-05-03:
Some downsides to consider:
All drawing to the screen needs to happen on the main thread, even when plugins create their own window(s).
Some plugins that use a lot of CPU are already heavily multi-threaded (our radar_pi uses one to two threads per radar).
Plugins (whether they are multi-threaded or not) will not probably exhibit interesting bugs when the plugin API calls methods from different threads.

jongough, 2018-05-03:
If you are dealing with external input that is not processed by the mainline then using your own threads is not a problem (probably). But when I run OCPN on a Raspberry PI with s63 charts and AIS going I see slowdowns when zooming or moving around. I am not sure what the answer is, but I do know that the PI shows a single CPU running flat chat and the others idle.

I think there are certain aspects that could/may be multi threadable. If you generate your own windows then you are responsible for them. If you are drawing on the main display then OCPN is in control and calls the  RenderOverlay and RenderGLOverlay functions within the plugin. These should be able to run on a different thread to the messaging system without any issues. The rendering on the mainline has to request a screen refresh if needed, but it cannot do it itself.

I am not sure how difficult it will be to do the multi threading, and I am sure if we attempt it there will be parts of the system that break. But that should not put us off.

taruti, 2018-05-05:
The codebase uses lots of global variables and adding more multithreading starts probably with clamping down on that.

jongough, 2018-05-05:
True, but we need to start somewhere and stop the increased use of globals and start to make non-readonly globals threadsafe, mutex's perhaps. I don't know which way to go, but I do think it needs doing, sooner rather than later, to make OCPN 'better' and more capable of being used on machines with less power per cpu (thread).

JonathanHuot, 2018-06-15:
Implementing multi-thread and improving performance of devices like RaspberryPi is a definitive **yes** !

closed 2021-04-13 by bdbcat

jongough, 2021-04-13:
Why was this closed? It is needed particularly for lower performance multi processor machines such as the pi. Plugins could be driven on their own thread when started by OCPN such that if it closed, for whatever reason, it can clean up.

bdbcat, 2021-04-14:
Jon...
It was closed as inactive and un-actionable as stated.
While an interesting subject, simply demanding multi-thread execution is not, IMHO, an "issue" for this venue.

jongough, 2021-04-14:
I raised an issue against pi-oesenc in flyspray, FS#2707 (https://opencpn.org/flyspray/index.php?do=details&task_id=2707), that multi-threading would help with. oesenc is taking so much processor that the system is slowing down on a PI 3B+, at times the UI appears to have stalled. I think there is an issue with oesenc and they way it keeps on loading and unloading charts, but that is another matter.

When in busy shipping areas with many AIS targets moving around and charts that keep on loading and unloading the 3B+ struggles to keep up. The 3B+ has 4 processors but only one is being used by the whole of OCPN. If OpenGL is turned on it does not actually help as whilst it uses another processor there is something strange as the processor usage seems to double but the UI slows down.

bdbcat, 2021-04-14:
Two points here:
1.  If oesenc is using too much CPU due to redundant loading of charts, that is a concern.  Thanks for the Flyspray issue. I will look at it.
2. If AIS processing is consuming too much CPU, that is another issue.  What we need to do is some profiling using probably an AIS target-rich VDR file, to identify where the bottlenecks might be.

It is fact, however, that all of the GUI actions must take place in a single thread, since none of the display managers we support is thread-safe.  So finding methods that a) take considerable time, and b) can be made thread-safe and efficiently communicate with the main GUI thread, is not so obvious.  Profiling is the first step...

jongough, 2021-04-14:
2. I think the issue is cumulative and the number of AIS tagets is the straw that broke the camels back. On my system I tend to run only a couple of plugins, oesenc for the latest official vector charts, logbook and OD. I do have wind instruments, and GPS information displayed by dashboard plugin when following a route. I have an AIS transceiver for GPS location and AIS information. I also am running SignalK to handle the usb ports and talk to OCPN, and OCPN sends out APB sentences to the autopilot through SignalK using the NMEA-NMEA plugin.

Unfortunately using SignalK getting a VDR file is not possible without adding the VDR plugin and that requires that SignalK send NMEA sentences to OCPN for it to record. I do have some VDR files, but they are not in heavy traffic areas or in areas where there are frequent oesenc chart changes. 

I thought all GUI stuff was done via OCPN initiating the 'draw' process and therefore giving the plugins their opportunity to draw on the screen. I don't think the plugins actually 'take' control without being offered it. For the other callbacks, i.e. NMEA stuff, these could be on a different thread and would not impact the GUI thread. I am only thinking out aloud, I don't know if this would be possible, but anything that takes load away from the GUI thread would be useful.

I know that OD uses timers to cause icons to flash and that these are on a separate thread, so if that is possible I would have thought other processes could be as well. OD can do some heavy handed processing through use of its API, i.e. weather routing and watchdog both make use of OD boundaries and test for objects, including AIS objects entering and being inside/outside boundaries. I don't really use this functionality, but it would also apply extra load to the main thread. I have tuned the API to try and cut down the processing as much as possible, but there can still be quite a few calls/sec and this is determined by the calling plugin.

rgleason, 2021-04-14:
Using[ TransmitterDan's vdrPlayer](https://github.com/transmitterdan/VDRplayer) to play big nmea AIS files such as the attached.  I also have a 10mb AIS file that took the CPU to 100% that I can upload to onedrive.
[Race-AIS-Sart-10m.zip](https://github.com/OpenCPN/OpenCPN/files/6311850/Race-AIS-Sart-10m.zip)

rgleason, 2021-04-14:
Onedrive shared link to  AIS  100percent CPU
https://1drv.ms/u/s!AojUASGm35WohJhoVQ8UM1b2oZKugQ?e=O07nix